wled/WLED #1954: Strip doesn't light up completely during the first effect loop

TimoUrban, 2021-05-09:
**Describe the bug**
When I use 'Lighthouse' with 'Analogous' as default effect after powering on the chip (after power loss), the strip is not lighted up in the first loop completely to the end.
In the following loops the strip lights up correct.

Because I'm using a smart plug to control my smart home devices with google assistant, this issue is showing up for me every time I power on the LED Strip.

**To Reproduce**
Save effect 'Lighthouse' and color 'Analogous' as a favourite. In config -  LED Preferences set this preset as default after powering on.
Power off the system. (remove power completely)
After you power on your board the default effect showing up but with the described issue.

**WLED version**
 - Board: Wemos D1 mini
 - Version 2104020
 - Format: binary

best wishes
Timo

Aircoookie, 2021-05-10:
Hi!
The way the Lighthouse effect works is that it "scans" across the strip and every LEDs it has not "scanned" yet remain off for the first loop. Additionally, WLED has a "fade up" effect on boot (or every brightness change indeed), which may cause the first few LEDs to remain off in the first loop despite being "scanned". You could disable the Crossfade transition or set its Transition time lower in LED settings, however this will of course affect every brightness change you do in WLED itself.

TimoUrban, 2021-05-10:
Hey!
Thank you for your detailed reply. I understand. 

Keep on going with this awesome project! :-)

closed 2021-09-14 by TimoUrban