wled/WLED #1955: unable to light up ws2812b led strip

D3OX, 2021-05-09:
Hey,
I have a  WS2812b led strip of 20 leds , powered by a 5v 10A supply.
i can't get the strip to light up. 

I've connected the Data Pin to D4 on my NodeMCU, along with GND, then GND & Power are connected seperately to the PSU.

Sending commands seems to work, but nothing lights up, no flickering, nothing. 
i've tried the color palette example from the fastled library

Any advice would be appreciated,

sbur83, 2021-05-09:
I had a similar thing happen with nodemcu ESP32 S, try D2 that worked for me.

D3OX, 2021-05-10:
I'll try that.

ventilator, 2021-05-29:
Especially NodeMCU pin D4 has to be high on boot. But the stripe sets it to low. So try to boot the NodeMCU and then connected the datapin. If this works, the solution is: solder a pullup resistor (4.7kohm between the datapin and 3.3V (not 5V). This should solve this specific problem, if you have it.

closed 2022-04-16 by stale[bot]