wled/WLED #1956: Not discoverable as ArtNet Node

aiotea, 2021-05-10:
I'm using [Lightkey](https://www.lightkeyapp.com/en/) for lighting control with ESP32 boards running the latest WLED, however it does not find any "artnet nodes" on my network.  

Lightkey is working with other artnet controllers, so it appears to be somthing with WLED. I believe it's a bug, but if theres a specific setting needed (besides enabling artnet in wled), please let me know

aiotea, 2021-05-11:
Thanks for the quick reply... They do have an option to add a specific IP, however I could not get that to work either. I selected artnet in wled, and then specified the IP and nothing happened.

closed 2021-09-19 by stale[bot]

forreal-periodt, 2023-02-16:
Hey @aiotea!

Wondering if you've had any success with this since then?
I'm looking to do a small project with a few addressable LED strips + WLED (ESP32) + Artnet.

Cheers!

MajorRedbeard, 2023-10-24:
I've been using WLED with Lightkey with the Preset mode, which works, though has flickering when adjusting the power (At least for my lights). The Effect mode seems to work a lot better, instant response and no flickering, but I'm having trouble getting the three (four with RGBWW SK6812) to map correctly when using LightKey. I can coerce the colours to be correct if I mess around with the RGB settings in WLED, but then all of the Palettes don't work because the colours have been shifted.

The ArtNet node discovery seems to work now, I'm using a Dig2Go (3 of them, actually), and it's able to find them on the network.
[image]