wled/WLED #1957: Drip langsamer machen

AchimFelsdrache, 2021-05-10:
was muss ich machen um Drip erheblich zuverlangsamen.

Und gibt es einen Weg alle Effects in einem Schlag zu verlangsamen
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2021-05-11:
Hi, soll die Fallgeschwindigkeit geringer sein oder die Anzahl/der Abstand zwischen den Tropfen?
Geschwindigkeit lässt sich mit dem Speed-Slider einstellen, die Tropfenanzahl mit Intensity. Beides auf 0 sieht schon sehr gut aus.
Ich werde den Wertebereich der Geschwindigkeit etwas anpassen, so dass ein noch langsamerer Fall möglich wird :)

Aircoookie, 2021-05-13:
Habe es angepasst, die Fallgeschwindigkeit ist nun bei geringster Geschwindigkeitseinstellungs nur noch halb so hoch.
Wenn du eigene Anpassungen machen möchtest, such einfach nach `SEGMENT.speed` in der jeweiligen Effektfunktion in `FX.cpp`. Der Wertebereich davon ist 0-255 und wird von der Effektfunktion individuell skaliert, sodass du mit einer Anpassung des Faktors viele Effekte relativ einfach langsamer bekommst :)

AchimFelsdrache, 2021-05-14:
Danke dir. Ich schaue mal ob ich hinbekomme. 

Gruss Achim

AchimFelsdrache, 2021-05-16:
Hi . 
Sei mir nicht böse aber um colortwinkels langsamer zu bekommen bin ich zu doof. 
Kannst du mir noch einen Tip geben bitte.

Gruss Achim

Aircoookie, 2021-05-17:
Bei Colortwinkles ist es in der Tat etwas komplizierter 😄 
Das Problem ist, dass die Pixel in jedem Animationsschritt (etwa 40 pro Sek.) ein wenig dimmen müssen, da sie sonst auf einer Farbe "stecken bleiben" können.
Als Workaround würde ich evtl. empfehlen, `return FRAMETIME` am Ende des Colortwinkle-Effekts durch `return FRAMETIME*3` zu ersetzen. Dies sollte die Geschwindigkeit auf ein Drittel reduzieren :)

AchimFelsdrache, 2021-05-17:
Werde ich versuchen Danke

closed 2021-09-26 by stale[bot]