wled/WLED #1958: Can't connect to wifi and power light's at same time

AndrewJLang, 2021-05-11:
Description:
I am using the nodeMCU esp8266 board.  I have successfully flashed the board with the latest stable release of the software, but encounter an issue when trying to connect to the network after 'rebooting' the device.  If I disconnect the power source from the device and wire the board to be connected to the lights, the WLED-AP network option will not appear anywhere (browser on PC, phone, etc).  If I leave the wiring to the LED's disocnnected and then just reboot the nodeMCU, I am able to successfully find the network and connect.  But as soon as I plug in the wiring for the lights again and check the connection, it is all offline.  What is the issue here, I thought it might be a network issue but I can connect as long as my lights are not connected to the board.

Reproduction Steps:
1) Flash software (latest WLED, version 12) onto the NodeMCU, see that it has successfully been flashed.
2) Unplug NodeMCU from computer usb which is giving it power.
3) Reconnect device to computer power source with no wiring for LED's connected to breadboard nodeMCU is resting on.  Successfully see the wifi option, input the wifi network name and password. Press save and connect.
4) Disconnect the device from computer USB power source and connect wiring for the power (5V 12A), D4 (output), and ground.
5) When powered on, blue light for device appears but nothing works within the WLED app and all devices appear offline.

Thank you for your help!

huggy-d1, 2021-05-11:
I think the **_bug_** is in the wiring.  Using USB power to power up the NodeMCU is fine.  However, for anything but a trivial small number of LEDs, USB does not provide enough power.

Normally a separate power supply (5V 12A in your case) is connected to the LED strips and the NodeMCU, with the 0V (common) wire from the supply connected to the NodeMCU GND and LED strip 0V wire.  
The 5V from the supply goes to the NodeMCU 5V and LED strip 5V wire.  
The NodeMCU data signal (D4 in your case) goes to the data or signal wire on the LED strip.

How many LEDs?  When the NodeMCU was on USB power, did you configure WLED with the # of LEDs, wifi credentials, anything else?

Hope this is helpful information.  Check the WLED wiki for examples for connecting things up.  It has pictures (and words) which may help you understand better what is happening.

AndrewJLang, 2021-05-13:
@huggy-d1 I am using a separate power supply as you specified.  I have the wiring done as you have described here.  I feel it is more of a software issue, why would the wiring of the lights (but still having the NodeMCU showing power) disable the wifi? How would the lights wiring interfere with the wifi signal if the wifi device is receiving power?

sansillusion, 2021-05-13:
Have you tried using a different ground pin on the MCU. I had an esp33 that
if I connected the ground to the one near the Vin it would not boot but
when I use the ground pin on the opposite side/corner it would work with no
issues.

Aircoookie, 2021-05-13:
Hi!
Not being able to boot once the LEDs are attached can also occur on the default pin D4 (GPIO2) if the LEDs pull the pin low, preventing the NodeMCU from booting. A few users have successfully mitigated this by placing an 330 ohm resistor across D4 and the 3.3V pin. Alternatively, you could try using TX (GPIO 1) as data pin (set LED pin 1 in settings).

AndrewJLang, 2021-05-17:
@Aircoookie I appreciate the suggestion, I received similar instructions from within the WLED discord regarding using a resistor across D4 and 3.3V pin.  My situation now is, when I add the resistor and I plug in my power supply, I will see the lights flicker for a moment when the power isn't all the way connected, then they will go out.  I will see the blue light on the NodeMCU light up, but when I go to find the WLED device via the mobile app, I cannot find it as there is seemingly no connection.  But, if I disconnect the wire running from my light strip into the D4 pin, I can suddenly discover the NodeMCU over Wifi, but then cannot communicate with the lights (obviously).  Another other suggestions into why this might be happening?  Power supply is 5V 15A.  It seems enough power is going to the lights for the initial flicker to occur, but not to the NodeMCU? Much appreciate the help

Aircoookie, 2021-05-17:
It is probably not an issue of too little power.
LED flicker when first connecting to power is very common, it is normal and shows your LEDs are correctly getting powered.
Just for testing sake, can you try using the TX pin (GPIO 1) as data pin?

AndrewJLang, 2021-05-17:
@Aircoookie connecting to the TX pin does not even turn light on for NodeMCU and no signal.  I tried connecting the resistor from TX from to 3.3V as well just for testing and still no results.  I will see the lights flicker so I am assuming they are getting enough power, but again cannot communicate with the app at all.  I'm also not sure what you meant by saying `set LED pin 1 in settings` from your first comment dealing with setting TX as the data pin.  I can't access any settings page. Any other suggestions?

AndrewJLang, 2021-05-18:
I have resolved the issue, the problem was with the LED's themselves and not the NodeMCU. The D4 connector wire to the LED's was broken, so connecting the wire coming out of D4 directly to the LED strip solved the problem. Thanks again for the suggestions.

closed 2021-05-18 by AndrewJLang