wled/WLED #1960: white flashes during animation with SK6812 12V stripes

newatschoen, 2021-05-13:
I have with a sk6812 12v stripes white flashes during animations...

I tested the version 0.12 stable and the 0.12.1-b1 version - but the issue stays :(

Aircoookie, 2021-05-13:
Hi! What is your setup? White flashes usually indicate that you might need a levelshifter: https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2

newatschoen, 2021-05-14:
Hi Aircookie

I can send a picture of the wiring if you want.

I will try a level shifter for my setup - will this one be fine?

ARCELI 10 stücke 4 Kanäle IIC I2C Logik Level Converter Bidirektionales Modul 3,3 V zu 5 V Shifter für Arduino https://www.amazon.de/dp/B07RDHR315/ref=cm_sw_r_cp_api_glt_i_0M6QV3VYJ3H9RPCESYWB?_encoding=UTF8&psc=1

Cheers
Sven

huggy-d1, 2021-05-14:
As a test, try limiting max current draw or max brightness until the flashing goes away.  If it never goes away, you may want to review the WLED wiki for wiring suggestions to see if you need to adjust your current data signal or common / GND sharing method.  If it does go away, you might still want to review that wiki while you wait on the level shifter.

newatschoen, 2021-05-14:
dear all - thx for your response. I have shortened the "green" wire ;) now it works fine :)

closed 2021-05-14 by newatschoen