wled/WLED #1962: Hyperion and WLED - WLED not connecting back to Hyperion after TV turned off

joshlovell15, 2021-05-16:
Hi everyone, 

As always, a wonderful development I love seeing progress. 

Over the weekend I had some spare time so I decided to play around with WLED for my Hyperion setup. 

So I had a spare NodeMCU V3 and decided to flash WLED 0.12.0 for control of my WS2812 LED strips behind my TV to relieve the Raspberry Pi of workload (mainly an excuse to go on another WLED adventure!) I used the same Raspberry Pi Zero W to host Hyperion, installed clean from HyperBian latest build. Configured WLED to my network with no issues regarding networking or WS2812 strip control. (WLED is tested working). Configured Hyperion to my network with no issues regarding networking or capture equipment (Hyperion is tested working). Configured Hyperion to point at WLED, WLED accepts Sync and connects to Hyperion (WS2812 display as they should with no lag/delay inbetween scenes - tested working).

The only issue I now face is that if I turn my TV off, I have No Signal Detection set on Hyperion to green as I have a "rainbow grabber" and the lights turn off after 4 seconds. However, turning the TV back on, I have to re-sync WLED to accept capture from Hyperion manually each time either via the WLED app or WLED webpage.

Can anyone please assist me with this issue, I've browsed the Wiki's for both developments and Google reveals it doesn't seem to be an issue for anyone else. I'm sure I'm probably missing a simple button or something. My WLED & Hyperion settings are attached below :) 

[image]
[image]
[image]
[image]

All help greatly appreciated, J

Aircoookie, 2021-05-17:
Hi! Thank you :)
How have you configured the connection to WLED inside Hyperion?
What steps do you have to follow to "re-sync WLED to accept capture from Hyperion manually each time either via the WLED app or WLED webpage"? 

Perhaps this https://github.com/Aircoookie/WLED/issues/1308 is related and your lights are turned off when turning off the TV/signal source, but not back on when the TV is turned back on?

joshlovell15, 2021-05-17:
Hi Cookie,

Thanks for the reply as I know you're a busy person :) 

I configured the connection in Hyperion using the following settings attached. 

[image]

I don't think my issue relates as everything works as intended and I do not want the lights to resume colours/effects before Hyperion was activated. My issue is that upon Hyperion sleeping it does not seem to reconnect to WLED automatically. 

Steps to resolve the issue after turning TV off and then back on are as follows;
- Turn TV on (check Hyperion for capture using Live Video window - source confirmed)
- browse to WLED webpage, power on WLED using toggle in top corner. Wait for Hyperion sync confirmation overlay window. 
- Hyperion and WLED then work as expected together. 

Upon typing this out, it appears that WLED may sleep after disconnection from Hyperion and that Hyperion doesn't wake WLED. Sorry for limited understanding :)

Again I appreciate your time

Aircoookie, 2021-05-17:
Alright, I do believe #1308 might be the issue you are experiencing. You could try if checking "Force max brightness" in the Realtime section of sync settings solves the problem :) Heads up, if you want to control the brightness of the ambilight, you'd have to do it inside Hyperion instead of using the WLED slider if you use this option.

joshlovell15, 2021-05-17:
I believe this has solved my issue but I'll report back after some testing, thank you for your ongoing support!!

closed 2021-05-18 by joshlovell15