wled/WLED #1963: Possible to use touch display for Configure Settings and showing Modes, wifi etc.

amitav2002, 2021-05-17:
Hi Team,

Thanks to develop such a great project. 

I have question related to use touch display with WLED.

- Can we use touch display for initial setting like configure wifi, change effect, other settings, everything what we do from WLED App.

- Show current wifi and selected effect on LCD.

I am having ILI9341 SPI TFT Screen.

Thanks
A.K. Singh

Aircoookie, 2021-05-17:
Hi!
Thank you :)
The WLED UI is built to run in a browser with HTTP/CSS/JS. It cannot be ported to run directly on the device.
That said, you could make a [usermod](https://github.com/Aircoookie/WLED/blob/master/usermods/EXAMPLE_v2/usermod_v2_example.h) to build an UI for the controls you want to be able to change and then have that set the WLED variables. Probably not feasible for adding every single setting and stuff, but e.g. changing the color could be doable

amitav2002, 2021-05-17:
Hi
Thanks very much for the response. 
I got it. 
Again, could you please help me, how we can enable the usermod. 
I am not sure about it. As I am new to it.
I appreciate, if you could advise how to enable "AUTO_SAVE" for an example.

Thanks
A.K. Singh

Aircoookie, 2021-05-17:
Hi!
A usermod for the display functionality you'd like does not yet exist.
Installing usermods works by copying the .h and .cpp files from the usermod folder to the `wled00` folder,
uncommenting the lines regarding that usermod in the usermods_list.cpp and compiling WLED.
I know it's not exactly easy, but I haven't found a better way yet :)

amitav2002, 2021-05-17:
Thanks much, I'll try for it.

closed 2021-09-26 by stale[bot]