wled/WLED #1964: Can I add the function of scanning nearby WIFI to select connection?

zx804262156, 2021-05-17:
Can I add the function of scanning nearby WIFI to select connection?
Because there are too many WIFI name symbols, it is difficult to fill in。

Aircoookie, 2021-05-17:
Hi!
That functionality is not currently supported, but if you'd like to add it yourself, I'd likely merge your pull request :)

huggy-d1, 2021-05-17:
If you use a PC to set up WLED the first time, you can easily type the symbols.  Of course, in these smart device times, not everyone has a PC available with wifi (so it can connect to the WLED AP and assign the SSID / password).

Another option which I find useful is to create a little "note" on my smart device with the wifi details.  Then I can copy/paste it into WLED fields pretty easily and without worrying I have not (temporarily) bricked WLED - until it automatically reverts back to AP mode.

sansillusion, 2021-05-17:
It would be great to see an easy wifi selection, would make it easier for
less tech savy people I make lamps for.

bad-jesus, 2022-03-18:
@laserguru 
Any progress on this? Would be interested to see how far you got if not.

laserguru, 2022-03-21:
Got sidetracked and never got to coding it, but I have made a sketch of the concept I had so far for the design of the interface, to share what little progress I got made.

Besides the main network selector, I think it's very important to have a simple and clear way two get back to the network selection mode (AP mode). My idea was to use the same GPIO line that needs to be pulled for flashing, because there's a user accessible button there most of the time.
There was a firmware, not sure which one (I think ESPeasy or Tasmota), that used the built in LED to indicate when you could press the button without triggering another boot mode. I think that can be implemented here too. That LED shares the pin with UART1 TX for DMX (UART0 is the one used for the bootloader), but if that gets initialized a little bit later, the LED can be flashed a couple of times to indicate you can press the button to get into AP mode. After that the startup would continue as normal.

blazoncek, 2022-03-22:
Default button (tagged button 0, defaults to GPIO0 but can be changed) already has the "start AP mode" built in. Just long press it for 5s-9s.

laserguru, 2022-03-22:
I just noticed that in button.cpp! I'm also going back the revision history and realized that was implemented **_ages_** ago! As in literally a couple of years. I had never noticed the feature! And it seems many users neither, judging by some tutorials I've come across that suggest switching off your router to invoke AP mode.

I think it would be the perfect moment to add a little text to this screen to explain how to use this function. I'll prepare a suggestion.

(And on a side note, what's the say on using the built in LED GPIO as an indicator?)

blazoncek, 2022-03-23:
There is a STATUSLED thing but has to be custom compiled.

PR #2895 (2022-11-24): feat(json): add wifi scanning
Adds wifi scanning to the JSON api at `/json/networks`. The initial request will trigger a scan, subsequent requests will scan or return the results depending on the state of the `WiFiScan`.

Add a `Scan` button next to the client ssid input, on click, scan for networks, and change the input to a select with the found ssids.

Fixes: #1964
commit 29b1f2af (2022-11-24): feat(json): add wifi scanning (#2895)

* feat(json): add wifi scanning

Adds wifi scanning to the JSON api at `/json/networks`. The initial
request will trigger a scan, subsequent requests will scan or return the
results depending on the state of the `WiFiScan`.

Add a `Scan` button next to the client ssid input, on click, scan for
networks, and change the input to a select with the found ssids.

Fixes: #1964

* Added option to go back to manual SSID input
closed 2022-11-24 by Aircoookie