wled/WLED #1966: Turning off Bus if off state causes random LED states

poelzi, 2021-05-18:
**Describe the bug**
#1245 causes a heavy regression on most of my uses of NodeMCU modules. Usually the first, sometimes more LEDs of a bus start to lit randomly. I always use RG174 cables to minimize interferenca and very good meanwell power supplies. The severity of this problem depends on the LED in use, SK6812 seem to be more sensitive, the WS2812 wire kind of strip is also super sensitive and starts to lit up daily.

Either we need a option to not stop the bus, if brightness is 0, or at least send a update very second or so.
The current state causes lots of problems.

**To Reproduce**
Depends on my factors, cable length, how dirty the mains are, etc. 
This kind of strip seems more sensitive: https://www.btf-lighting.com/products/usb-rgb-led-pixels-string-christmas-lights-2m-5m-10m-rf-21keys-controller-ws2812b-ic-addressable-individually-light-10-leds-m-5v

**Expected behavior**
LEDs should be off when state is off

**WLED version**
 - Board: NodeMCU
 - Version v3
 - Format both

huggy-d1, 2021-05-18:
I was just talking with a friend about WLED and a similar problem.  Mine has to be noise because when the LEDs have lit, WLED has been in a "power off" state even though the ESP8266 and LED strip are powered up waiting for a signal.

In the "power off" case, it would be nice if WLED periodically reaffirmed brightness of 0, and when in a solid color state, have an option to reaffirm color and brightness periodically, perhaps even a slider (scaled 0 to 100%, with tic marks associated with easy points to update).

poelzi, 2021-05-18:
@huggy-d1 you can test my PR if it fixes your problem

blazoncek, 2021-05-19:
If leds are randomly turning on after WLED has been turned off, it may be caused by a noisy data signal line.
Try adding capacitors to the data line or other noise filters.
You can also use a WLED board like @srg74 WLED Wemos shield that employs a relay to cut power off from LED strip itself.
You can test if #1245 is the cause of your problems by using pin (GPIO) other than the one with onboard LED (usually GPIO2).
This is not a SW issue.

poelzi, 2021-05-25:
I see radom LED states on different controllers, esp32 with RMT and nodemcu. It is a combination from a software and hardware issue. It is not that I can easily change this installation, it is quite large with lots of leds. Multiple 25A power rails spanning many meters.
I'm not the only person having this problem. Especially the string light ws2811 and SK6812 leds seem to be much more sensitive to noise. I have seen random LED states on nearly all strips I installed after some days/weeks or hours, depending on the installation. We could reduce the framerate to power thos of, but refreshing this state is not a wrong solution.

blazoncek, 2021-05-25:
I can agree that updating bus with _black_ color can be a remedy for occasional blinking but that is just a _band-aid on a bleeding wound_. The wound will not stop bleeding. Your LEDs **will** still blink alas for the shorter time. Perhaps unnoticeable, perhaps very noticeable still.
And while speaking of _black_ color, why not creating a preset with #000000 (black) color and Solid effect? It will server the same purpose as far as turning off LEDs go. It will maintain full FPS though.

And again: If you are indeed using "multiple 25A power rails" it is even more reasons to implement switching power off for the LED strip (using relays on the AC side of PSUs). This would completely alleviate your (and other's) problems.

blazoncek, 2021-10-11:
Forcing bus refresh even in off state has been added to latest beta.
Feel free to reopen if it doesn't fix your problem.

closed 2021-10-11 by blazoncek