wled/WLED #1971: Invert single pin of PWM driven rgb light

Tenn0, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
My problem is, that on my current controller, one pin needs to inverted to show the correct color on the whole strip. If its at the standart setting, red is when set to 0 at 255 brightness and at 0 brightness when turned to 100% red.

**Describe the solution you'd like**
In the light setup, make every pin in a pwm driven led itself invertible, not just all pins at once.

**Describe alternatives you've considered**
Switching to ESPHome, but that doesnt support e131 for pwm lights (yet). 

Thank you for your ideas for making WLED better!

fishbone-git, 2021-11-29:
I would also like to see this.  Now that Athom is offering pre-loaded bulbs with WLED, there is a case where one of their other sizes needs to invert PWM5.  Even though 5CH Colour temperature it not yet supported, it doe s create unexepcted results with the pin set up with teh wrong logic.

illuxions, 2021-12-22:
I have 4 each of the 15w and 7w Anthem bulbs and ran in to the issue of needing to inverter only the cold temperature line.  Is there any quick and dirty I can do?  I've been rummaging through the code looking at the PWM stuff but I'm at a loss, I'm sure the Lagavulin isn't helping but I'm stuck.

sansillusion, 2021-12-22:
Just a suggestion, mabe use a mixture of NPN and PNP MOSFETs ?

illuxions, 2021-12-24:
> Just a suggestion, mabe use a mixture of NPN and PNP MOSFETs ?

WLED already has the ability to invert all PWM signals, I was trying to figure out which module handles it but still haven't found it.

It's done in software in Tasmota, any PWM out can be set as inverted.  

I'm not sure why Anthem used different CCT drivers between bulbs but the 7w has the cold channel regular and the 15w has the cold channel inverted.  I've found how the inverting is handled in Tasmota but still haven't figured out how it's done in WLED.  If all channels can be inverted it should be easy enough to just invert a single channel.

blazoncek, 2021-12-24:
It is trivial from technical point but will take a lot of time to develop necessary UI.

bjhiltbrand, 2023-01-24:
This is not only an issue with Athom bulbs, but also with Feit Electric 6in. RGBW Recessed Downlight (LEDR6/RGBW/AG) modules. Tasmota has the pinout listed [here](https://templates.blakadder.com/feit_electric-LEDR6_RGBW_AG.html) with GPIO13 mapped to PWM5 inverted. None of the other pins are inverted. When I click the inverted checkbox, it currently kills the colors and just sets all the leds to full bright. I assume this is because only 1 pin should be inverted. With the inverted checkbox unchecked, everything works as it should, with exception of the cool white end of the temperature slider. As I slide more toward the cool white end, the warm white leds dim and go out, but the cool white leds never come on as they are supposed to.

bjhiltbrand, 2023-01-24:
> It is trivial from technical point but will take a lot of time to develop necessary UI.

Why do you have to develop the UI too much? Couldn't you just have the UI accept a negative number in the GPIO field? The presence of a negative value would mean you want to invert whatever pin specified. In the case of the Feit bulb mentioned above, GPIO 5 is the issue. A value of -5 could indicate an inverted GPIO 5. Just a thought.

softhack007, 2023-01-25:
> A value of -5 could indicate an inverted GPIO 5. Just a thought.

Actually its not possible this way. Any GPIO value <0 means "pin not assigned", which is used - everywhere in WLED - to disable the driver. 

A possible solution would be to add a new driver "PWM with one inverted channel" - however this would be "something special" (Germans call it Extra-Wurst) only for your controller, so chances are not very high as there are a  number more important issues / challenges we currently have to resolve ...

bjhiltbrand, 2023-01-27:
> Actually its not possible this way. Any GPIO value <0 means "pin not assigned", which is used - everywhere in WLED - to disable the driver temporarily. Also your proposal would fail for GPIO 0, which is a valid pin on ESP32. "-0" = 0.

Ah, good to know. I can clone the repo and see if I can figure out something here, but my C++ skills are very rough. Would you be able to point me to where the existing drivers (especially the PWM+CCT driver) are written, so I can study up on them and understand how they work?

blazoncek, 2023-01-27:
> Would you be able to point me to where the existing drivers (especially the PWM+CCT driver) are written, so I can study up on them and understand how they work?

bus_manager.h

serhat-ozkara, 2023-08-16:
This issue holds true for other devices also. I flashed a Xiaomi Philips CCT light which needs this. Other one is a feican RGBW lightbulb.