wled/WLED #1973: Changing LED strip pin continues to output pattern on all old pins until AC cycle

nolanhergert, 2021-05-19:
**Describe the bug**
After changing the pin field in "Hardware Setup" under LED preferences, the previous pin is still outputs information and isn't torn down, even after a reboot. Only after an AC cycle are the previous pins "decommissioned"

**To Reproduce**
I kept the first pin set, but added and changed the 2nd pin used. It was first set to 12, then 25, then 32, then 1, then 3. All of them continued to output data.

**Expected behavior**
The previous pin no longer outputs LED strip data when the pin is changed.

**WLED version**
 - Board: ESP32 dev
 - Version 0.12.1-b1
 - Format Binary

**Additional context**
This is an awesome framework and saves me a ton of time! Donating now and hopefully more later.

Thank you for your help!

blazoncek, 2021-05-19:
I think this is a known bug in NeoPixelLibrary for ESP32 or at least the way WLED uses that library.
There is little WLED can do about it.
You can use a workaround to re-use old GPIO pins in new strips added and then delete them without actually outputting anything on them.

nolanhergert, 2021-05-19:
Not a big deal for me. Thanks for the info! Closing.

closed 2021-05-19 by nolanhergert