wled/WLED #1974: Palettes Become null When 11 Additional Segments Are Added

leoherzog, 2021-05-19:
**Describe the bug**
On a fresh boot of WLED 0.12.0, if I add 11 additional segments via UI or API, the `palettes` key becomes `null` and the Web UI throws errors.

**To Reproduce**
Boot WLED, create 11 additional segments. In my case, I have 60 LEDs and each segment is 5 in length.

Fresh boot:
[image]

**WLED version**
 - Board: [ESP8266](https://www.amazon.com/gp/product/B07HF44GBT)
 - Version: 0.12.0
 - Format: Binary

**Additional context**

Fresh boot state:
```javascript
{"state":{"on":true,"bri":128,"transition":7,"ps":-1,"pl":-1,"ccnf":{"min":1,"max":5,"time":12},"nl":{"on":false,"dur":60,"fade":true,"mode":1,"tbri":0,"rem":-1},"udpn":{"send":false,"recv":true},"lor":0,"mainseg":0,"seg":[{"id":0,"start":0,"stop":60,"len":60,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false}]},"info":{"ver":"0.12.0","vid":2104020,"leds":{"count":60,"rgbw":true,"wv":true,"pin":[2],"pwr":681,"fps":2,"maxpwr":850,"maxseg":12,"seglock":false},"str":false,"name":"tiempo2","udpport":21324,"live":false,"lm":"","lip":"","ws":0,"fxcount":118,"palcount":56,"wifi":{"bssid":"redacted","rssi":-25,"signal":100,"channel":11},"fs":{"u":32,"t":1024,"pmt":0},"ndc":0,"arch":"esp8266","core":"2_7_4_7","lwip":1,"freeheap":23048,"uptime":7,"opt":127,"brand":"WLED","product":"FOSS","mac":"redacted"},"effects":[
"Solid","Blink","Breathe","Wipe","Wipe Random","Random Colors","Sweep","Dynamic","Colorloop","Rainbow",
"Scan","Scan Dual","Fade","Theater","Theater Rainbow","Running","Saw","Twinkle","Dissolve","Dissolve Rnd",
"Sparkle","Sparkle Dark","Sparkle+","Strobe","Strobe Rainbow","Strobe Mega","Blink Rainbow","Android","Chase","Chase Random",
"Chase Rainbow","Chase Flash","Chase Flash Rnd","Rainbow Runner","Colorful","Traffic Light","Sweep Random","Running 2","Aurora","Stream",
"Scanner","Lighthouse","Fireworks","Rain","Tetrix","Fire Flicker","Gradient","Loading","Police","Police All",
"Two Dots","Two Areas","Circus","Halloween","Tri Chase","Tri Wipe","Tri Fade","Lightning","ICU","Multi Comet",
"Scanner Dual","Stream 2","Oscillate","Pride 2015","Juggle","Palette","Fire 2012","Colorwaves","Bpm","Fill Noise",
"Noise 1","Noise 2","Noise 3","Noise 4","Colortwinkles","Lake","Meteor","Meteor Smooth","Railway","Ripple",
"Twinklefox","Twinklecat","Halloween Eyes","Solid Pattern","Solid Pattern Tri","Spots","Spots Fade","Glitter","Candle","Fireworks Starburst",
"Fireworks 1D","Bouncing Balls","Sinelon","Sinelon Dual","Sinelon Rainbow","Popcorn","Drip","Plasma","Percent","Ripple Rainbow",
"Heartbeat","Pacifica","Candle Multi", "Solid Glitter","Sunrise","Phased","Twinkleup","Noise Pal", "Sine","Phased Noise",
"Flow","Chunchun","Dancing Shadows","Washing Machine","Candy Cane","Blends","TV Simulator","Dynamic Smooth"
],"palettes":[
"Default","* Random Cycle","* Color 1","* Colors 1&2","* Color Gradient","* Colors Only","Party","Cloud","Lava","Ocean",
"Forest","Rainbow","Rainbow Bands","Sunset","Rivendell","Breeze","Red & Blue","Yellowout","Analogous","Splash",
"Pastel","Sunset 2","Beech","Vintage","Departure","Landscape","Beach","Sherbet","Hult","Hult 64",
"Drywet","Jul","Grintage","Rewhi","Tertiary","Fire","Icefire","Cyane","Light Pink","Autumn",
"Magenta","Magred","Yelmag","Yelblu","Orange & Teal","Tiamat","April Night","Orangery","C9","Sakura",
"Aurora","Atlantica","C9 2","C9 New","Temperature","Aurora 2"
]}
```
Immediately after I go to the "Effects" tab and add 11 more segments; 0-5, 5-10, 10-15, etc:
```javascript
{"state":{"on":true,"bri":128,"transition":7,"ps":-1,"pl":-1,"ccnf":{"min":1,"max":5,"time":12},"nl":{"on":false,"dur":60,"fade":true,"mode":1,"tbri":0,"rem":-1},"udpn":{"send":false,"recv":true},"lor":0,"mainseg":0,"seg":[{"id":0,"start":0,"stop":5,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":5,"stop":10,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[102,153,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":2,"start":10,"stop":15,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,204,51,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":3,"start":15,"stop":20,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,51,204,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":4,"start":20,"stop":25,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[102,0,153,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":5,"start":25,"stop":30,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":6,"start":30,"stop":35,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[105,150,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":7,"start":35,"stop":40,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,207,48,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":8,"start":40,"stop":45,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,54,201,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":9,"start":45,"stop":50,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[99,0,156,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":10,"start":50,"stop":55,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[252,0,3,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":11,"start":55,"stop":60,"len":5,"grp":1,"spc":0,"on":true,"bri":255,"col":[[108,147,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false}]},"info":{"ver":"0.12.0","vid":2104020,"leds":{"count":60,"rgbw":true,"wv":true,"pin":[2],"pwr":397,"fps":2,"maxpwr":850,"maxseg":12,"seglock":false},"str":false,"name":"tiempo2","udpport":21324,"live":false,"lm":"","lip":"","ws":0,"fxcount":118,"palcount":56,"wifi":{"bssid":"redacted","rssi":-23,"signal":100,"channel":11},"fs":{"u":32,"t":1024,"pmt":1621442001},"ndc":0,"arch":"esp8266","core":"2_7_4_7","lwip":1,"freeheap":22496,"uptime":163,"opt":127,"brand":"WLED","product":"FOSS","mac":"redacted"},"effects":[
"Solid","Blink","Breathe","Wipe","Wipe Random","Random Colors","Sweep","Dynamic","Colorloop","Rainbow",
"Scan","Scan Dual","Fade","Theater","Theater Rainbow","Running","Saw","Twinkle","Dissolve","Dissolve Rnd",
"Sparkle","Sparkle Dark","Sparkle+","Strobe","Strobe Rainbow","Strobe Mega","Blink Rainbow","Android","Chase","Chase Random",
"Chase Rainbow","Chase Flash","Chase Flash Rnd","Rainbow Runner","Colorful","Traffic Light","Sweep Random","Running 2","Aurora","Stream",
"Scanner","Lighthouse","Fireworks","Rain","Tetrix","Fire Flicker","Gradient","Loading","Police","Police All",
"Two Dots","Two Areas","Circus","Halloween","Tri Chase","Tri Wipe","Tri Fade","Lightning","ICU","Multi Comet",
"Scanner Dual","Stream 2","Oscillate","Pride 2015","Juggle","Palette","Fire 2012","Colorwaves","Bpm","Fill Noise",
"Noise 1","Noise 2","Noise 3","Noise 4","Colortwinkles","Lake","Meteor","Meteor Smooth","Railway","Ripple",
"Twinklefox","Twinklecat","Halloween Eyes","Solid Pattern","Solid Pattern Tri","Spots","Spots Fade","Glitter","Candle","Fireworks Starburst",
"Fireworks 1D","Bouncing Balls","Sinelon","Sinelon Dual","Sinelon Rainbow","Popcorn","Drip","Plasma","Percent","Ripple Rainbow",
"Heartbeat","Pacifica","Candle Multi", "Solid Glitter","Sunrise","Phased","Twinkleup","Noise Pal", "Sine","Phased Noise",
"Flow","Chunchun","Dancing Shadows","Washing Machine","Candy Cane","Blends","TV Simulator","Dynamic Smooth"
],"palettes":null}
```

blazoncek, 2021-05-19:
The solution is already available in my fork, dev branch.
May be incorporated into master at some (not so far) point.

This is a limitation of ESP8266 and can be slightly mitigated by disabling some features during compile.

leoherzog, 2021-05-25:
Confirmed that this issue is not present on the ESP32.

leoherzog, 2021-09-27:
This issue is still present on the ESP8266

blazoncek, 2021-09-27:
@leoherzog which version are you talking about?

@Aircoookie this will most likely need a JSON buffer 10k in size if @leoherzog is using the latest code. ArduinoJson Assistant calculates 10000 bytes minimum size (but we have mitigated nested col arrays, so perhaps a bit less is needed).

closed 2022-04-16 by Aircoookie

kudlav, 2022-05-14:
This issue is still present on the ESP8266 with v0.13.1

kudlav, 2022-05-14:
I've added the following line at the begining of `populatePalettes` as a workaround and seems ok now.

```js
function populatePalettes(palettes)
{
	if (palettes === null) palettes = [];
```

igorsantos07, 2024-08-10:
This seems valid for my ESP8266 with only 5 segments... But I guess, given the feeling it's a very edgy case, it won't be fixed?

blazoncek, 2024-08-10:
@igorsantos07 You are missing key information. The underlying issue has been fixed.