wled/WLED #1975: set/get sleep timer via JSON

CWempe, 2021-05-19:
**Is your feature request related to a problem? Please describe.**

I would like to be able to de/activate the timer via JSON.
And read the time left (in minutes?).

**Describe the solution you'd like**

I guess a json property could look like this:

```json
{	
  "state": {
    "timer": false
  }
}
```

Maybe `"timer": true` could be replaced by `"timer": 45` to indicate the time left.
But maybe a separate property would be better.

huggy-d1, 2021-05-20:
Something like this work?  Seems like a reasonable non-breaking json update to me.
```
{	
  "state": {
    "timer": false
    "time_remaining": 45
  }
}
```

CWempe, 2021-05-20:
Looks good.

How about a property to set a custom timer?

Like: `"timer" : 45` to set a 45 minute timer.

Or make `time_remaining": 45` settable. 🤔

CWempe, 2021-08-02:
Any chance this could be part of v13.0.0? 😇

blazoncek, 2023-08-05:
You can set the nightlight mode (timer) using "nl" JSON API.
[KB](https://kno.wled.ge/interfaces/json-api/)

If this is not what you need please re-open.

closed 2023-08-05 by blazoncek