wled/WLED #1978: Connection problems with the Node MCU ESP8266 on ver. 0.12.0

oOJoshOo, 2021-05-20:
The connection to my router cannot be established with version 0.12.0, switching off the WiFi sleep mode does not help.
With the 0.11.1 version everything works fine, only the new functions are missing.

**WLED version**
 - Board: [Node MCU  ESP8266]
 - Version [WLED_0.12.0_ESP8266]
 - Format [Binary]

huggy-d1, 2021-05-21:
Try a full erase before loading 0.12.x.  Also, try to use a newer 0.12.x version.

fennsen64, 2021-05-26:
I've found that wled seems to have a general issue with the NodeMCU modules. e.g. Sometimes it takes minutes before it get's a dhcp address after several wifi restarts.
In addition, the wifi connection strenghts is _always_ much poorer than in comparison with a Wemos D1 Mini (tested several NodeMCU modules which works flawless with tasmota or something else) in same place on same AP. So I migrated all wled-controllers to D1 Minis right now.

oOJoshOo, 2021-05-27:
> Try a full erase before loading 0.12.x. Also, try to use a newer 0.12.x version.

After I tried an downgrade to version 0.11.1 again via ota ubgrades , I did that and it worked flawlessly. Thank you 

There seems to be a problem with ota upgades on the nodemcu

closed 2021-05-27 by oOJoshOo