wled/WLED #1981: SyntaxError: Unexpected token in JSON at position 3063

kashkashx, 2021-05-21:
**Describe the bug**
When opening the WLED app or after sometime while using the WLED app, a red error box will appear notifying that an error has occured (exact text in title name). 

This error causes you to lose all of your saved presets without any way to recover them.

**To Reproduce**
It seems to happen intermitently, but has been happening to me consistently over the past day.
Has not happened until today and the app has been perfectly functional for roughly two weeks. 

**Expected behavior**
I should recieve no error and my presets should not all be deleted. 

**WLED version**
 - Board: ESP-WROOM-32 ESP32 ESP-32S Development Board (Brand Aitrip)
 - Version: 0.12.0
 - Format [](url)

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

Aircoookie, 2021-05-23:
Hi!
Sorry! This is not the first time this problem occured, there must be a problem in the file handling.
If you have not yet reset the presets, you can send me your `[WLED-IP]/presets.json` file and I can at least recover your presets for you.

blakewenloe, 2021-09-22:
Just adding more info that may help identify the problem. This happened to me yesterday and today after overwriting some presets. It removed the ID of the preset but left the rest of the configuration.

Example:

This is how it should have looked for preset 3:
```
  "3": {
    "n": "Back Door Red",
    "on": true,
    ...
```

This is how it saved the file:
```
  {
    "n": "Back Door Red",
    "on": true,
    ...
```

senwerks, 2022-04-30:
Just chiming in that this is an ongoing issue for me too, and seems to happen when you copy/paste presets to new entries. I was copying presets from one WLED install to a few others, and it corrupted/wiped all of them. I keep backups of the presets, so not a huge deal, but figured I should still report it's still an issue.

blazoncek, 2022-04-30:
A bit more data about your HW and SW set-up would be beneficial.
Please post your cfg.json and presets.json (both available from the security settings page as a backup or via /edit page).

jordandrako, 2022-05-01:
I have this issue quite often, but today I tried to save the first preset of a new installation.

### HW/SW:

Board: Quinled ESP32
WLED version: 0.13.1
Build: 2203150
LEDs: WS21812b Fairy lights 635 count

### Corrupted presets.json (problem at column 407):
``` json
{"1":{"n":"Incandescent","on":true,"bri":145,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":615,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Tree","col":[[188,255,5],[0,0,0],[0,0,0]],"fx":0,"sx":21,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":615,"stop":635,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Star","col":[[255�158,13],[0,0,0],[0,0,0]],"fx":102,"sx":144,"ix":179,"pal":0,"sel":false,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}}
```

It's the `"col":[[255�158,13],[0,0,0],[0,0,0]]` bit.
Fixed json:
``` json
{"1":{"n":"Incandescent","on":true,"bri":145,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":615,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Tree","col":[[188,255,5],[0,0,0],[0,0,0]],"fx":0,"sx":21,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":615,"stop":635,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Star","col":[[255,158,13],[0,0,0],[0,0,0]],"fx":102,"sx":144,"ix":179,"pal":0,"sel":false,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}}
```

blazoncek, 2022-05-01:
Can anyone else pinpoint the error as @jordandrako did?

The ‘"col":[...]‘ content is constructed manually...
If others have the same point of error then this may be the case of a code bug.

blazoncek, 2023-08-05:
The root cause for preset corruption is still being investigated (is unknown) but the workaround has been put into place in 0.14.
Closing. Feel free to re-open if the problem persist by providing additional (debug) data.

closed 2023-08-05 by blazoncek