wled/WLED #1984: Show EffectID and PaletteID on MouseOver or when clicked on the GUI

jkwim, 2021-05-22:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
When the Web GUI is used, if the EffectID and PaletteID can be shown with MouseOver or when clicked, that would help for people like me who try to send MQTT commands from another device using JSON API.

Thank you for your ideas for making WLED better!

blazoncek, 2023-05-26:
Effects (with the visualization) are listed in [KB](https://kno.wled.ge/features/effects/).
Similarly palettes are [listed](https://kno.wled.ge/features/palettes/) too.

Adding "title" to buttons to display IDs may seem good to technical users but may be confusing for others.

I propose to close this issue without implementing.

blazoncek, 2023-06-07:
Fixed in #3220