wled/WLED #1985: TypeError: Cannot read property 'name' of undefined

memphisraynz, 2021-05-23:
Today I have just started getting this above error when controlling my WLED setup.

I am using an ESP8266 with the stand ESP8266(4mb) firmware.
I was using 12.b5 and had issues upgrading via OTA so i had to erase and overwrite with firmware v12.
I Mainly control the LED's with Home Assistant, I am currently on version 2021.5.5. Last update was done about a week ago.
The lights get used on a daily basis for my son's wake up and nightlight. Now which is failing.
As well as the above error I am also getting "Cannot read property of 'success' null"

Nothing new has happened today or in the last few days, so I am a little stumped on what has gone wrong. They last worked as expected a few hours ago, while my boy was turning the lights on. 

Any help is appreciated and if I can provide any more information, will be happy to assist.

Aircoookie, 2021-05-23:
Hi! I'm very sorry about that. 
How many LEDs is your controller running?
This error seems to indicate that the JSON API response to a query by the UI was faulty. You can try opening `http://[WLED-IP]/json/si` in your browser and sending me what it outputs. It's possible that this happens due to a low memory condition, does the problem persist after a reboot?

memphisraynz, 2021-05-23:
Hi yes, after a reboot the same behavior happens. I am  running 100 leds.
I went to that url, changing to my IP, and I only get 'null'

huggy-d1, 2021-05-24:
Ahh, thank you to Notepad++ for the json viewer!
Try one of the later releases of 0.12.x available here: https://github.com/Aircoookie/WLED/releases

closed 2022-04-16 by stale[bot]