wled/WLED #1989: "Realtime" UDP is pretty slow

sixtemesseven, 2021-05-23:
I'd like to use a 300 LED WS2811 strip as audio spectrometer.

https://github.com/sixtemesseven/AudioSpectrumStrip

The fft and udp transfer is pretty fast, loop runs at about 100hz. However, the ESP32 connected to the WLED is really slow (in the few hz range).
Is there a way to speed up the update frequency on the WLED side? Or would the DMX protocoll be a better way to go?

I have a custom Arduino Firmware for the UDP which runs at maybe 10-15hz but I was looking for a. easier setup and b. faster update rates.

Thx :)

huggy-d1, 2021-05-24:
Have you considered the WLED fork with sound reactive built in?  Either as a fork, or as the solution?  Seems like it is doing something differently because there have been no real complaints that I'm aware of with update rate.

sixtemesseven, 2021-06-05:
I will look into it. Many thanks :)

closed 2021-06-05 by sixtemesseven

LennyPenny, 2021-06-08:
@sixtemesseven how did you solve your issue? 
does https://github.com/atuline/WLED really have better UDP performance? 

I also want to send full frames to my ~300 pixel led strips at ~120 fps.

sixtemesseven, 2021-07-26:
> @sixtemesseven how did you solve your issue?
> does https://github.com/atuline/WLED really have better UDP performance?
> 
> I also want to send full frames to my ~300 pixel led strips at ~120 fps.

I have to do a direct comparison. Give me a few days.

xurei, 2024-02-18:
I'm trying to use WLED 0.14.0 with the realtime UDP and I have the same observation : LEDs are blinking in an unpredictable way, some frames are dropped or shown for less than 100ms while I'm sending frames at a contant rate of 2Hz (verified to be stable on the PC side)

@sixtemesseven did you succeed to make it run at a rate of at least 10Hz ? If so, how did you manage to make it work.

For context : I'm using 4 LED strips in parallel with 150 LEDs each on an ESP32.