wled/WLED #1992: If Blynk enabled, LED stopped to flash.

amitav2002, 2021-05-25:
**Describe the bug**
Dear Team,

Thanks much for develop such great WLED. When I came to know about it, I returned my pre-purchased pixel controller. :-)

Here , I would like to bring one issue to you, I have recompiled source code after enable the Blynk. 
Now I am facing some kind of issue after configured the Blynk under sync interface. All works perfectly if blynk Server is reachable. Anyhow it Internet is not available and Blynk server unreachable, my led has stopped to flash. As soon as if I make blynk server reachable, again it's started to work properly.

Please advise, If I am missing any thing when compiling. Do I have to make any changes in order to fix this.

**WLED version**
 - Board: [ESP32 dev]
 - Version [0.12.0]
 - Format [self-compiled]

Thank you for your help!

Aircoookie, 2021-05-25:
Hi! I'm glad you like it :)
This is normal, if a Blynk server + token is configured, but that server is not reachable, the integration tries to reconnect, which causes the LEDs to freeze for 5 seconds very frequently. To use Blynk, you need to ensure that the server is reachable.
I don't think that is changeable on my end, but I will try to update the Blynk library to test if the behavior changed.

amitav2002, 2021-05-25:
Thanks very much for the response.

That's ok if it's normal behavior with Blynk Configuration.

Professionally, I really don't know to write code. Even then, I am little curious to know, is there any way that we could configure the polling for blynk server for specific time, if Server not available then WLED go with it's own functionality instead freeze. 

And in background continue poll/ping blynk server, once it's available, then it consider blynk functionality as well.
:-)

Thanks Again.

anrada73, 2021-05-29:
I did it this way. if the Internet is not available, then after the restart, the system turns on its access point and at the same time the LEDs do not try to connect to inaccessible servers (blynk,time,mqtt etc.)

amitav2002, 2021-05-31:
Oh! I got it, Thanks.

closed 2022-04-16 by stale[bot]