wled/WLED #1993: Can't set colors via JSON API in first segment when segment 2 is powered off

muffin142, 2021-05-25:
**Describtion**
With 2 segements ans segement 2 powered off i can't set the first segments color via JSON API:

{"seg":{"id":0,"i":[[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0]]}}

With segement 2 powered on it works well.

**To Reproduce**
I set up an WLED Node with 30 Pixel WS2811 LED-Strip and created 2 Segments.
- Segment 1: START LED: 0, STOP LED 10, GROUPING 1, SPACING 1
- Segment 2: START LED: 20, STOP LED 25, GROUPING 1, SPACING 1

With both Segments powered on in the segments section and the whole device powered on I can set the segements colors vie JSON API:
{"seg":{"id":0,"i":[[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0]]}}
{"seg":{"id":1,"i":[[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0]]}}

With section 2 powered off and section 1 powered on and configured to solid color, section 1 will turn off when sending

{"seg":{"id":0,"i":[[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0],[0,255,0],[0,0,255],[255,0,0]]}}

The other way round it works fine (section 1 powered of, section 2 powered on and sending colors as shown obove for section 2)

**Expected behavior**
Setting section 1 color whatever the power state opf section 2 is.

**WLED version**
 - Board: ESP8285
 - Version: 0.12.0 "hikari"
 - Format: Binary, 2M Version

huggy-d1, 2021-05-25:
What do you do to "power on" and "power off" a segment?

blazoncek, 2021-05-25:
> What do you do to "power on" and "power off" a segment?

There is a button in UI for that. ;)
Expand the segment and you'll see it.

Jojopirat, 2021-12-09:
This problem has only existed since version 0.12.0. In version 0.11.0 the deactivation of individual segments still worked perfectly.
powering on individual segments is managed by gui or sending following json {"seg":{"id":1,"on":true}}

closed 2022-04-27 by stale[bot]