wled/WLED #1994: Support Overlapping Segments

leoherzog, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
If I have two segments whose LEDs overlap, they should elegantly combine their lit colors instead of blinking and flashing.

**Describe the solution you'd like**
If I were to have 10 LEDs (`0-9`) with two segments: `0-9` set to `rgbw(0,255,0,0)` and `4-6` set to `rgbw(255,0,0,0)`, the resulting LEDs would be:
```javascript
[
  {
    "col":[0,255,0,0]
  },
  {
    "col":[0,255,0,0]
  },
  {
    "col":[0,255,0,0]
  },
  {
    "col":[0,255,0,0]
  },
  {
    "col":[255,255,0,0]
  },
  {
    "col":[255,255,0,0]
  },
  {
    "col":[255,255,0,0]
  },
  {
    "col":[0,255,0,0]
  },
  {
    "col":[0,255,0,0]
  },
  {
    "col":[0,255,0,0]
  }
]
```

I realize this could get complicated when it comes to combining effects, but that sure would be nice as well. For example, if segment `0` in the above example was set to `Solid` and segment `1` was set to `Breathe`, then it would breathe between `rgbw(0,255,0,0)` and `rgbw(255,255,0,0)` as if segment `1` was layered on top of `0`.

Thanks!

blazoncek, 2021-05-25:
You can achieve the same breathing effect by setting secondary color to the same color as first segment primary color.

Overlapping segments are difficult to implement properly for every available effect, palette and combination of speed and intensity.

huggy-d1, 2021-05-26:
@Aircoookie fantastic idea!  Each effect would get a new attribute or property that states whether that effect supports overlapping.  In the UI, somehow allow filtering of effects that allow overlapping, or note it with a symbol (which then requires a legend).  Probably a better way to visualize it.  And of course, need a new json / http api request: Request Effects that support overlapping.  This way, someone could generate random but acceptable overlapping segments and effects via script and send to wled to act on.

GargamelCZ, 2022-02-05:
This would help me so much - I'm installing a long strim into my living room and I want to control parts of it separatedly for different situations, but it would be so cool to once in a while to set it all to a single long effect. How deep is this in your backlog?

GargamelCZ, 2022-02-05:
Btw. I've got two other solutions to this problem:
1) Add a "layout id" to a segment. Then for each layout you can define separate set of segments. You can have one layout with segmets for let's say each wall in a room. And a second layout with only one segment all the way. Etc.

2) Much simpler solution problably: Segmets, that are "turned off" won't "blink black" or just once, when you turn them off, but after that they will ignore if another segment is writing over them.

Aircoookie, 2022-02-06:
@GargamelCZ for your use case specifically, just make a preset containing all your segments, and another with just a single segment, encompassing the entire sign.
This exactly matches your first idea of "layouts" - essentially every preset in WLED can be a different "layout". 
Presets support different segment configuration for a reason :) At least for your use case there is no need for overlapping segment support (which does not mean it is not required or useful for other purposes)

blazoncek, 2023-03-08:
Successfully overlapping segments depends on the used effects. Some allow previous values of LEDs to show and some don't.
Recent commits in 0.14.0-b2 added the possibility to use certain effects in "overlay" mode allowing overlapping segments.

closed 2023-03-08 by blazoncek