wled/WLED #1996: The brightness slider is doing almost nothing when controlling DMX lights

BN3000, 2021-05-27:
**Describe the bug**
The brightness slider is doing almost nothing when controlling DMX lights.
I have a DMX controllable LED strip and this works fine with WLED.
I can change the colors and effects but I have one problem.
When I want to adjust the brightness with the brightness slider, the brightness barely changes.
The minimum brightness stays at a reasonable high brightness.
The difference between the max brightness and the minimum brightness is negligible.

To see where the problem is I also added a small WS2812 LED strip to my WLED setup so I can compare the DMX LED strip with the WS2812 LED strip.
When I then adjust the brightness with the brightness slider, the brightness of the DMX LED strip barely changes but the brightness of the WS2812 changes over the whole range of the slider and can also set to a verry low brightness.

I also tried some other DMX lights but with the same result.

**To Reproduce**
1. Connect a DMX controllable LED strip and a WS2812 LED strip to the WLED controller.
2. Select a solid color.
3. Change the brightness level (from maximum to minimum and back)
4. You will notice that the brightness of the DMX controllable LED strip barely changes and that the brighness range of the WS2812 LED strip is good.

**Expected behavior**
The brightness of DMX controllable light can be adjusted more accurate over the whole range. A this moment the minimum brightness level for DMX devices is also to high.

**WLED version**
 - Board: ESP-12E (ESP8266)
 - Version: Build 2105171
 - Format: self-compiled to have a version with DMX functionality (compiled with Visual Studio Code)

BN3000, 2021-05-28:
I tested this by using the WLED webinterface GUI and also tested this via the home assistant integration. Both give the same result.

When I test the DMX controllable LED strip with another DMX controller I can use the whole brightness range. I even can go so low that the LEDs are OFF (what is the normal behavior). I used an OLA server on a raspberry pi together with an USB to RS485 dongle to test this.

I followed this instructions [https://github.com/Aircoookie/WLED/wiki/Compiling-WLED](url) and [https://github.com/Aircoookie/WLED/wiki/DMX-Output](url) to compile this WLED build (WLED shows build version 2105171). I compiled WLED for the D1mini.

BN3000, 2021-09-27:
Little tap to keep the ticket open.

BN3000, 2022-04-16:
Little kick to keep the ticket open.

Aircoookie, 2022-04-16:
Just a hunch, but this might be due to your DMX light's response to the brightness channel and whether it needs gamma correction or not.
One thing you can try is replacing `uint8_t brightness = strip.getBrightness();` in dmx.cpp line 19 with `uint8_t brightness = briT;` before compiling. If this doesn't help, try `uint8_t brightness = strip.gamma8(briT);`

lplassman, 2022-05-06:
I think this issue is fixed. I created a PR that fixes the brightness when no shutter DMX channel is set. @BN3000 would you mind checking out the latest main branch and see if your issue is fixed?

BN3000, 2022-06-04:
I created the same setup as before with the latest main branch and this is working finen now!

closed 2022-06-04 by BN3000