wled/WLED #1998: Can connect but not change any effects (ESP32)

DSew20, 2021-05-28:
Hello:

**Describe the bug**
I am new to your products and can't wait to get them running.

Am using the dig-uno board to drive LED.  For testing, using a small 5 LED module (5v, WS2812B) and powered up.
I can connect to WLED.  But, no matter what I do in the settings all is see is 1 blue LED light up.
I have set LED count to 5, LED Output to SW281x, 0..5, on Pin 16.  Connected the LED data cable to the Uno LED 1.

It worked once and I could change the settings and effects.

Now, I can log on to the interface, but can not changes any LED effects.  I just get a single, static blue LED no matter what I do.

**To Reproduce**
As above.  I can connect to WLED.  But, can not get WLED to change the effects.

**Expected behavior**
The LED changes to the chosed effect settings.

**WLED version**
 - Board: ESP32, on a QuinLed Dig-Uno
 - Version: WLED 0.12.1-b1_ESP32
 - Format: Binary

**Additional context**

Thank you for your help!

huggy-d1, 2021-05-28:
What version dig-uno do you have?  V3?
Have you measured the voltage at the LED strip +5V pad (compared to the 0V/GND pad)?

DSew20, 2021-05-28:
Thank you.  Voltages seem ok.

I have re-flashed V0.12.0, set-up Wifi and so far its ok.  Have restarted a few times and came back on with LED working as expected.  Not sure if there was an issue with the earlier flash or something else.  But, re-flashing V012.0 seems stable at the moment.

blazoncek, 2021-05-31:
If issue is resolved, please close it.

closed 2022-04-16 by stale[bot]