wled/WLED #2000: Needed help controlling leds, effects from backend code

Saliksheraz, 2021-05-31:
Thanks a lot for this awesome application.
I am a very beginner in coding wled, I just want to create a custom effect that simply combines a couple of effects and runs them after a specific time, but I want to code them and use an API to turn it on/off so that user don't need to create a preset by his own self each time he uploads bin file. I'm using wled00/usermod.cpp file. 
server.on("/openDoorEffect", HTTP_GET, [](AsyncWebServerRequest *request) {
              WS2812FX::Segment &seg = strip.getSegment(strip.getMainSegmentId());
              seg.mode = FX_MODE_COLORTWINKLE;
              request->send_P(200, "text/plain", "true");});
I am having 2 segments, im able to apply an effect on main segment by this code.
I just want the simplest way to change effects/speed/intensity change color of effects, and someway to set a timer between these events,
Thanks again :)

blazoncek, 2021-05-31:
You can do that using presets and palylists (within presets).
No need to code.
Flash content is only lost if you erase flash by means of attaching ESP to PC using serial. If you upgrade OTA all settings are preserved.

Saliksheraz, 2021-05-31:
But actually, I want to make a bin file and send it to my customers so that they just upload the bin on their ESPs without needing to make presets.

blazoncek, 2021-05-31:
To make your customers happy you'll need to put some effort in: research a bit, learn a bit, code a bit, test a bit and deploy a bit. ;)

Saliksheraz, 2021-05-31:
Yeah bro that's why I'm coding, researching, asking, testing, etc :)

huggy-d1, 2021-06-01:
You have customers and no product yet.  That's fantastic.

closed 2021-12-18 by blazoncek