wled/WLED #2003: After restarting my Android phone, I now only have intermittent connectivity to my WLED devices through the app or client IPs. I still can access through the access point

HecticSkeptic, 2021-05-31:
**Describe the bug**
I am able to ping my WLED devices from my laptop that is connected to the same WiFi network but I am not able to ping them from my phone.

**To Reproduce**
I cannot seem to reliably reproduce the issue because it is intermittent. More often than not, I cannot reach my WLED devices from mobile. I get ERR_ADDRESS_UNREACHABLE  

**Expected behavior**
A few days ago I was not having this issue. It seemed to start after restarting my phone but I cannot be sure. It definitely seems that the issue is with my mobile device and not something else, but perhaps it is my wifi settings or something.

**WLED version**
 - Board: ESP32 
 - Version: 0.12.0
 - Format: Binary

**Additional context**
I have Xfinity and I am not the account holder. My router gateway gives me basically no control over my wifi settings. Our 2.4 and 5 ghz bands have the same SSID and I cannot separate them. This hasn't been an issue in the past though, and does not prevent my laptop on the 5ghz band from pinging my WLED devices on the 2.4 GHz band. 

Changing some of my VPN settings on my phone yesterday seemed to fix the issue for a short period but the issues quickly returned so I think that was just a coincidence. In general, I have my VPN off and have disabled it auto starting or really doing anything 

Thank you for your help!

HecticSkeptic, 2021-05-31:
To add, my friend's IPhone cannot discover the lights either. So it does not seem to be an issue that is unique to my Android device. But is rather an issue with any mobile connecting to the WLED devices via my home WiFi

sansillusion, 2021-05-31:
Have you tried connecting to the wled device using IP in browser. If it
works it may be that your router is blocking MDNS/Bonjour discovery service.

HecticSkeptic, 2021-05-31:
the IP in browser access works on PC but not on mobile.  so it's not just the app that has issues

HecticSkeptic, 2021-05-31:
result of pinging my phone from my laptop

[image]

HecticSkeptic, 2021-05-31:
result of pinging my WLED devices from my laptop

[image]

HecticSkeptic, 2021-05-31:
Attempt to ping laptop (10.0.0.4) and WLED esp32 (10.0.0.186) from my android phone:

[image]
[image]

sansillusion, 2021-05-31:
That is weird. I would mabe have tough an "isolate devices" option in the
router but it works from laptop to phone but not vice versa so probably not
that... I can only guess it's a phone thing, mabe some antivirus/firewall
app in your phone but you said that your friend's iPhone also can't connect
to wled. There is something actively prevent communication between phone
and wled it's most likely a "security" option but what it is I have no clue.

HecticSkeptic, 2021-05-31:
> That is weird. I would mabe have tough an "isolate devices" option in the router but it works from laptop to phone but not vice versa so probably not that... I can only guess it's a phone thing, [...]

Yeah the fact that this is happening across multiple mobile devices is what's really tripping me up.  Also not sure why I can't even seem to ping my laptop from mobile..... or why the laptop tried to ping itself

blazoncek, 2021-06-01:
Definitely not a WLED issue but router/access point _feature_.

HecticSkeptic, 2021-06-01:
> Definitely not a WLED issue but router/access point _feature_.

Any idea which _"feature"_ is causing this issue?

blazoncek, 2021-06-01:
I can't say for your setup but most likely some type of "guest" network policies or "client isolation" policies.

closed 2022-04-16 by stale[bot]