wled/WLED #2005: VL53L0X_gestures Usermod unintentional switching

Esquilo007, 2021-06-01:
Hi everyone!
*********************
My Setup:
- WEMOS D1 Mini
- WS2812b's
- VL53L0X TOF Sensor for switching on/off and adjusting the brightness according  to usermod VL53L0X_gestures
- WLED 0.12.0, version shortly after the usermod was released
*********************
My Problem:
There is nothing else in my setup. I'm using the usermod "as it is", made no changes at all.
VL53L0X - Sensor is supplied via the 3.3V of the WEMOS D1 Mini (measured and stable).
The Sensor itself also works, switching on/off via wiping hands as well as adjusting the brightness, the only problem is that often, but unregularly, it switches on or off the lights unintentionally, even when there is no movement.
Maybe anyone else has the same problem?
Or maybe anyone with the animated staircase usermod?
Or maybe anyone knows, if there is a way to adjust the sensitiveness if that is the problem.

Maybe the Usermod Autor @eg321 also has a suggestion / idea 
(by the way, thank you very much for this great usermod! I want to use it in the kitchen, where it is awesome when you don't need to touch something)

Thank you all in advance!
Simon

eg321, 2021-06-03:
@Esquilo007 thanks for your feedback! Glad to hear that it useful for someone else.

I'm using same sensor for ~6 month and there were no unintentional switching, but please note that it can react for the steam (my teapot was too close with sensor).
Also you can check VL53L0X_MAX_RANGE_MM parameter (230 mm by default). It defines max height that sensor reacts for gestures. You should make it smaller, if you have some accessories below the sensor:
https://github.com/Aircoookie/WLED/blob/master/usermods/VL53L0X_gestures/usermod_vl53l0x_gestures.h#L25

Esquilo007, 2021-06-03:
Hi @eg321,
Thank you for your reply.
I think your mod is a very good possibility to control the LEDs.
In the discord forum someone also gave me the tip to mount the sensor straight pointing 90 degrees downwards.
Right now it is mounted at an angle of roundabout 40 degrees downwards.
Maybe this is a problem, I'll try.
What is the angle you've got yours mounted?
Thank you in advance.

eg321, 2021-06-04:
@Esquilo007, In my case sensor is mounted at ~55 cm above tabletop pointing 90 degrees downwards. Not sure if it's rootcause, but you can check..

closed 2022-04-16 by stale[bot]