wled/WLED #2012: WLED Usermod own Website

Bebick, 2021-06-03:
Hey guys,
i tried to start my own site from the usermod. Unfortunately without success....
How can I call my own website from a usermod?
Or do I have the possibility to implement my own website via the Usermods button? If so what do I have to consider?

Greetings

pbolduc, 2021-08-20:
You want to make an outbound http request? Redirect to your site? Not really clear what you are trying to do.

blazoncek, 2021-08-20:
Do you want your usermod to _generate_ a dynamic HTML page or just display a static HTML?
If it is the later, you can just upload your HTML files (and CSS or JS) using /edit page.
If it is the former you will need to hook into WLED web server using `server.on(...);`

closed 2022-04-16 by Aircoookie