wled/WLED #2013: Current BTF-LIGHTING WS2815 12 Volt?

jolichter, 2021-06-03:
Hello, I would like to share my measurement here. I get only 40 watts although this strip is specified with 90 watts. Can someone explain this to me possibly?

Technical data from [btf-lighting.com](https://www.btf-lighting.com/products/1-ws2815-dc12v-led-pixels-strip-light-dual-signal): Power: 0.3W/LED  x 300 LEDs = 90 Watt

To reproduce:
- AZDelivery D1 Mini NodeMcu with ESP8266-12F
- Firmware WLED_0.12.0_ESP8266
- BTF-LIGHTING WS2815 Modell BTF-12V-060L-W (5m / 300LEDs)
- Power supply MW LPV-150-12 (12VDC / max. 10A)
- Power supply converter 12DC to 5 VDC for the ESP (ground connected to 12 volt power supply)
- all LEDs on (Brightness 100%): win&SM=0&S2=300&RV=0&SB=255&A=255&R=255&G=255&B=255&R2=0&G2=0&B2=0&FX=0&T=1
- Multimeter Fluke 115 True RMS

Sketch:
[image]

WLED current limit 880mA, WLED estimate 866mA, measured with Fluke 780mA:
[image]

WLED current limit 1500mA, WLED estimate 1488mA, measured with Fluke 1290mA:
[image]

WLED current limit 6000mA (means open without limit), WLED estimate 4000mA, measured with Fluke 3300mA:
[image]

The last measurement is equal to 40 watts and the voltage of the power supply remains stable at exactly 12.0 volts. To me the LEDs are bright enough and everything works very well, but why are the power specifications not correct? Does anyone have any tips or ideas?

Thanks!

huggy-d1, 2021-06-03:
I suggest you visit https://quinled.info and review the spreadsheet of power / voltage / led strip style and compare all the types of LED strips @Quindor has tested.
Then see if you could develop a simple formula or table that would apply to the majority of LED strips of a given brand, keeping in mind the reason for the current limit and current calculations is to prevent an overheating situation that may lead to fire and sad days recovering from a fire at home.

Hopefully by understanding why they may _intentionally_ read differently will make your scientific study for your specific LED strips even more useful.

You are also welcome to download WLED source code, modify the calculations so they are more precise for your LED strips, and use your own custom WLED firmware.

jolichter, 2021-06-04:
Thanks, primarily I wonder why the max 90W is not reached. To reach the 90W, a current of 25mA per LED needs to flow. If I understand it correctly, I would have to create a fork or ask for a feature request to set the parameters from 12mA to 25mA to reach max. 90 watts? I guess I need a new setting: "WS2815 (25mA)". If someone wants to prevent overheating, can simply leave the maximum current at 4A, for example.

fishbone-git, 2021-06-04:
The software isn't a closed loop and isn't current controlled.  The 12mA is an estimate and I believe it was made based on one persons measurements.  It doesn't mean that your LEDs are being limited just becuase it reports a lower value than what is really happening as the function only caps the brightness to set the estimated current under the limit you set as a very last resort way of allowing use of an under powered supply.  From a safety point of view, it's not a good way to prevent overheating and certainly shouldn't replace careful design.

jolichter, 2021-06-04:
Oh damn, I checked the link from huggy-d1 and now I realize it: WS2815 is not BTF2815 !
The BTF2815 is a WS2815 clone which uses a UCS1903 chip.
Here are the links: [quinled.info](https://quinled.info/2020/03/12/digital-led-power-usage/) and [YT Movie](https://www.youtube.com/watch?v=PbVQceESH9s)

Thanks!