wled/WLED #2014: WS2812

dexer12, 2021-06-03:
Hi,

i really couldn't find anything regarding this anywhere :(
As i got it already, there seems to be two kind of WS2812 LEDs.
Some of them are RGB some of them are GRB. Now i build my own Nanoleafs just to fine out i mixed up these two types of LEDs.
There is an option in Hardware Setup to add multiple LED Stripes, but just on different PINs.

Is it someway possible to configure both "different" LEDs on the same PIN?
So 1,2,3,4 on PIN 2 RGB 5,6,7,8 on PIN 2 GRB.

thanks in advanced!

huggy-d1, 2021-06-03:
After you add the 2nd strip to the end of the first strip, you configure the 2nd segment to have a different color order.
You cannot connect 2 wires to pin 2 and get what you want if that is what you are asking.
The LED commands need to move along the 1st LED strip, then at the end, the data signal needs to go to the 2nd LED strip.  If you need to run a wire all the way back to the ESP8266 so they can start together and have different color order, then that is what you need to do.  Or, use another pin, or another ESP/WLED.

Aircoookie, 2021-06-03:
Correct, selecting a different color order for two chained strips on the same pin is unfortunately not an easy option.
In addition to the two options given by @huggy-d1 , it is possible to have two different color orders on the same pin with a custom compilation with some build flags. `-D COLOR_ORDER_OVERRIDE -D COO_MAX=35 -D COO_MIN=0 -D COO_ORDER=0` would force the first 35 LEDs to use GRB order and the others whatever is set in the UI. If you don't want to set up VS Code, let me know your platform (ESP8266/ESP32) and how many GRB and RGB LEDs you have :)

dexer12, 2021-06-04:
Thank you both for you quick answers and help!

i have printed 11 "nanoleafs" so its usually one LED Strip with all 81 LEDs all in all. but they are soldered on a lot of spots.
I didnt think about checking the LEDs before soldering, expected they were all the same. Now i have LED 1 RGB, 2 - 8 GRB, 9 - x RGB then another 2 GRB and so on. So they are all mixed over.

I created a second segment in the segments section, but where could i setup that they are a different color order? if this is a possible way, it should be fine i think. And a lot easier, unsolder and resolder with other LEDs again.
The only option i found to setup the color Order is in the Hardware Setup. 

@Aircoookie 
Its and Wemos Mini D1 so a ESP8266 right? But the LEDs are totally mixed up, so really thanks for your help. But thats a lot of playing around. I think i will do try it on my own, if thats the last option.

Aircoookie, 2021-06-04:
If there are totally mixed up, that of course makes it difficult... That override only works if there is one zone of LEDs with a different color order :(

For segments, there is no dedicated option to set the color order, using them is only a workaround (e.g. if you want Red, you set the color to green)

dexer12, 2021-06-04:
Okay, thank you both a lot for you help and your time!
So i'll go and find our which LEDs to change and solder it again.
I'm not going to do this mistake again in the Future!

dexer12, 2021-06-07:
I got it working :)

i read and played alot. But this post was helpful:

https://github.com/Aircoookie/WLED/issues/1052

I have to edit a lot of items:

`void SetPixelColor(uint16_t indexPixel, RgbwColor c)
  {
    RgbwColor col;

    uint8_t co = _colorOrder;
    if (indexPixel >= 0 && indexPixel < 1) co = COL_ORDER_RGB; //SEGMENT 1
	  if (indexPixel >= 1 && indexPixel < 9) co = COL_ORDER_GRB;
	
	  if (indexPixel >= 9 && indexPixel < 10) co = COL_ORDER_GRB; //SEGMENT 2
	  if (indexPixel >= 10 && indexPixel < 12) co = COL_ORDER_RGB;
	  if (indexPixel >= 12 && indexPixel < 13) co = COL_ORDER_GRB;
	  if (indexPixel >= 13 && indexPixel < 15) co = COL_ORDER_RGB;
	  if (indexPixel >= 15 && indexPixel < 16) co = COL_ORDER_GRB;
	  if (indexPixel >= 16 && indexPixel < 18) co = COL_ORDER_RGB;
	
	  if (indexPixel >= 18 && indexPixel < 28) co = COL_ORDER_RGB; //SEGMNT 3
	
	  if (indexPixel >= 27 && indexPixel < 36) co = COL_ORDER_RGB; //SEGMNT 4
	
	  if (indexPixel >= 36 && indexPixel < 45) co = COL_ORDER_RGB; //SEGMNT 5
	
	  if (indexPixel >= 45 && indexPixel < 46) co = COL_ORDER_GRB; //SEGMNT 6
          if (indexPixel >= 46 && indexPixel < 48) co = COL_ORDER_RGB;
          if (indexPixel >= 48 && indexPixel < 49) co = COL_ORDER_GRB;
          if (indexPixel >= 49 && indexPixel < 51) co = COL_ORDER_RGB;
          if (indexPixel >= 51 && indexPixel < 54) co = COL_ORDER_GRB;
			
	  if (indexPixel >= 54 && indexPixel < 63) co = COL_ORDER_GRB; //SEGMNT 7
	  if (indexPixel >= 63 && indexPixel < 72) co = COL_ORDER_GRB; //SEGMNT 8
	
	  if (indexPixel >= 72 && indexPixel < 73) co = COL_ORDER_GRB; //SEGMNT 9
          if (indexPixel >= 73 && indexPixel < 75) co = COL_ORDER_RGB;
	  if (indexPixel >= 75 && indexPixel < 79) co = COL_ORDER_GRB;
	  if (indexPixel >= 79 && indexPixel < 81) co = COL_ORDER_RGB;
	`

But now every single wrong LED is working like it should. And all effects could be used :)
There maybe a more perfomance way to solve this. But i'm not doing much else with the wemos d1 then managing theses 81 LEDs.

Aircoookie, 2021-06-07:
That's adventurous, I'm very happy that you got it working without having to re-do the entire hardware :)

closed 2021-06-07 by Aircoookie

dbassy, 2023-11-23:
Sorry where about in the WLED app in the web browser would you write this code to change the colour order if you have 2 different types of WS2812 strips?

blazoncek, 2023-11-23:
@dbassy that is a built-in feature of WLED since #2463.
Available in LED settings.