wled/WLED #2016: JSON Playlist:  Repeat Indefinitely

pmf2021, 2021-06-04:
When I send in a 0 for repeat, my lights do nothing.  When I send in any other value, it works.  I found that if I send in a "-1", it cycles indefinitely.  The code must be decrementing down to zero.  

Quick fix:  Update documentation so that indefinite is -1, not zero.

WLED_0.12.0_ESP8266_2M.bin

p.s.  Thank you for such wonderful software!!!!!

blazoncek, 2021-06-07:
-1 is indefinite & random.
0 is supposed to be just indefinite (maintaining the order of playlist entries).
But as you found out, there is a bug that prevents playlist from running in such case.

blazoncek, 2021-06-07:
As a temporary workaround you can set a high number as a "repeat" property.
Unfortunately this is limited to 127 ATM, I already prepared a fix and will submit a PR to @Aircoookie

Aircoookie, 2021-06-07:
Fix merged to master :)

closed 2021-06-07 by Aircoookie