wled/WLED #2017: wLED on ESP-12f Crashing

nstiesi, 2021-06-05:
New user here. Just installed wLED on ESP-12f to drive a Ws2811 strip. Installation and setup went well and I just could not stop grinning as I was using it. However once I walked away I kept coming back to the default condition.

I tried bunch of stuff related to power supply, but nothing has worked. I thought at first the power was dipping, but I dont think that was it. I decided to monitor the serial port for exceptions and turned up the below every time it resets. I am not a coder by nature, so none of this means much to me, but I was hoping someone could help me figure this out? I have incorporated wLED into a larger project and it is almost done, I just need to push this over the line. Thanks!

22:14:14.557 → Exception (0):
22:14:14.557 → epc1=0x402a149c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
22:14:14.557 →
22:14:14.557 → >>>stack>>>
22:14:14.557 →
22:14:14.557 → ctx: cont
22:14:14.557 → sp: 3ffffb20 end: 3fffffc0 offset: 0190
22:14:14.557 → 3ffffcb0: 3ffea2ce 00000000 00000000 4024b6e0
22:14:14.557 → 3ffffcc0: 00000000 3fff3684 00000010 3ffef0a0
22:14:14.557 → 3ffffcd0: 00000002 00000000 4026f9b0 00000000
22:14:14.557 → 3ffffce0: 00000000 00000000 00000000 3ffef054
22:14:14.557 → 3ffffcf0: 4029f7c0 4026f9b0 00000000 00000000
22:14:14.557 → 3ffffd00: 00000000 00000000 00000000 4026fa93
22:14:14.557 → 3ffffd10: 00000002 00000000 4026f9b0 00000000
22:14:14.557 → 3ffffd20: 00000000 00000000 00000000 40105876
22:14:14.557 → 3ffffd30: 40105858 3fffc100 3fffdad0 00000000
22:14:14.557 → 3ffffd40: 00000000 4026f9b0 3fff20fc 401022b8
22:14:14.557 → 3ffffd50: 400005e1 4010472e 3ffee700 00040000
22:14:14.557 → 3ffffd60: 4026f9b0 00000033 00000010 40103030
22:14:14.557 → 3ffffd70: 40105876 04000102 00000000 00000001
22:14:14.557 → 3ffffd80: fbf8ffff 04000002 3feffe00 00000100
22:14:14.557 → 3ffffd90: 0000001a 00000018 04000102 40105858
22:14:14.557 → 3ffffda0: 3fffc100 3fffdad0 0000001c 00000000
22:14:14.557 → 3ffffdb0: 40104b9f 00000000 0000001f 40100968
22:14:14.557 → 3ffffdc0: 00000000 00000000 0000001f 40100968
22:14:14.557 → 3ffffdd0: 40104667 0000007f 3fffc228 40105e11
22:14:14.557 → 3ffffde0: 4000050c 00000030 00000010 ffffffff
22:14:14.557 → 3ffffdf0: 4000dd24 00000030 00000010 ffffffff
22:14:14.557 → 3ffffe00: 4000066d 00178812 4bc6a7f0 00000017
22:14:14.557 → 3ffffe10: 00004bc6 f3bf4ee0 00000000 fffffffe
22:14:14.557 → 3ffffe20: 00002855 3fffc6fc 59434ee0 00178812
22:14:14.557 → 3ffffe30: 4bc6a7f0 00178812 00000000 00000030
22:14:14.557 → 3ffffe40: 40100a05 00000000 3fff0f3c 00000000
22:14:14.557 → 3ffffe50: 40105870 4010586d 00000000 00000041
22:14:14.557 → 3ffffe60: 400005e1 00000003 00000000 3ffefeaa
22:14:14.557 → 3ffffe70: 40216cbe 00000030 00000010 00000030
22:14:14.557 → 3ffffe80: 4022eaf9 00000000 00000000 00000000
22:14:14.557 → 3ffffe90: 00004bc6 00000000 00000000 fffffffe
22:14:14.557 → 3ffffea0: 00000000 3fffc6fc 00000000 3ffefeaa
22:14:14.557 → 3ffffeb0: 00000000 3fffdad0 3fff0ef8 0000001c
22:14:14.557 → 3ffffec0: 0000002b 00000000 40100a60 00000000
22:14:14.557 → 3ffffed0: 00000000 4bc6a7f0 d374bc6a 00000000
22:14:14.557 → 3ffffee0: 00000000 00000000 4bc6a7f0 00000000
22:14:14.557 → 3ffffef0: 00000000 4bc6a7f0 e4189374 00000000
22:14:14.557 → 3fffff00: 00000000 4bc6a7f0 e9ba5e35 00000000
22:14:14.557 → 3fffff10: 00000000 00000000 4bc6a7f0 00000000
22:14:14.557 → 3fffff20: 00000000 00000000 40100a60 00000611
22:14:14.557 → 3fffff30: 00000000 5619ff83 3ffefe00 4022c344
22:14:14.557 → 3fffff40: 000704b6 00000000 00000001 40100968
22:14:14.557 → 3fffff50: 3ffffeb0 3fff0dac 3ffefeaa 3fff0ef8
22:14:14.557 → 3fffff60: 3fffdad0 00000000 3fff09d8 40217e5b
22:14:14.557 → 3fffff70: 3fffdad0 00000000 3ffefeaa 4022eaf6
22:14:14.557 → 3fffff80: 00000000 00000000 00000001 3fff0ef8
22:14:14.557 → 3fffff90: 3fffdad0 00000000 3fff0eb8 4022f674
22:14:14.557 → 3fffffa0: feefeffe feefeffe feefeffe 4024b1c4
22:14:14.557 → 3fffffb0: feefeffe feefeffe 3ffe8830 401008cd
22:14:14.557 → <<<stack<<<

The exception decoder says:

Exception 0: Illegal instruction PC: 0x402a149c EXCVADDR: 0x00000000 Decoding stack results 0x40100968: realloc(void, size_t)* at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\ umm_malloc.cpp line 1116 0x40100968: realloc(void, size_t)* at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\ umm_malloc.cpp line 1116 0x40100a60: schedule_recurrent_function_us(std::function const&, unsigned int, std::function const&) at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/ PolledTimeout.h line 129 0x40100a60: schedule_recurrent_function_us(std::function const&, unsigned int, std::function const&) at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/ PolledTimeout.h line 129 0x40100968: realloc(void, size_t)* at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\ umm_malloc.cpp line 1116 0x401008cd: realloc(void, size_t)* at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\ umm_malloc.cpp line 1090

**To Reproduce**
It is random, but crashes usually come in waves.

**WLED version**
 - Board: ESP8266 2_7_4_7
 - Version: wLED 0.12.0 build 2104020
 - Format: Binary

**Additional context**
I have tried different power supplies.  I have also tried a complete flash erase which seemed to help for a while, but ultimate the crashes came back with slightly different error codes

09:43:12.099 -> Exception (28):
09:43:12.099 -> epc1=0x40216cbe epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
09:43:12.099 -> 
09:43:12.099 -> >>>stack>>>
09:43:12.099 -> 
09:43:12.099 -> ctx: cont
09:43:12.099 -> sp: 3ffffdd0 end: 3fffffc0 offset: 0190
09:43:12.099 -> 3fffff60:  3fffdad0 00000000 3fff09d8 40217e5b  
09:43:12.099 -> 3fffff70:  3fffdad0 00000000 3ffefeaa 4022eaf6  
09:43:12.099 -> 3fffff80:  00000000 00000000 00000001 3fff0ef8  
09:43:12.099 -> 3fffff90:  3fffdad0 00000000 3fff0eb8 4022f674  
09:43:12.099 -> 3fffffa0:  feefeffe feefeffe feefeffe 4024b1c4  
09:43:12.099 -> 3fffffb0:  feefeffe feefeffe 3ffe8830 401008cd  
09:43:12.099 -> <<<stack<<<
09:43:12.099 -> 
09:43:12.099 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40216cbe
EXCVADDR: 0x00000000

Decoding stack results
0x40217e5b: tcp_split_unsent_seg at core/tcp_out.c line 986
0x401008cd: umm_init() at C:\Users\nick.stiesi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 475

Thank you for your help!

Aircoookie, 2021-06-05:
Hi!
Sorry for the crashes! 
How many LEDs are you running on what pin, and does this happen on one specific effect or can you reproduce it on different effects?
Exception 0 (Illegal instruction) is quite rare, it could be a problem with the flash memory. Do you have a second identical board to try?

nstiesi, 2021-06-05:
I have 134 total leds in 44 groups of 3.  Its a 12v ws2811 based strip that was cut to this length.  The strip is being powered with 12V direct from an AC-DC (5A) supply.  This supply powers a 2 channel CNC controller as well.  The CNC controller has an on board 5V regulator, which feeds a 3.3V regulator that feeds the ESP.

The data is going through the default pin (gpio2, I believe).  The crashes do not appear to be effect based.  Its quite random.  I had over 2 hours of uptime today, this after 5-10 crashes in the first 20 minutes of power on.  Then one random crash, now I am back going on 45 minutes as of this writing.

I do have another esp-12f.  I have these mounted to a custom dev board, and I broke a tool while cutting a new board for to try the other ESP, which ruined the last of my copper clad material.  I ordered more, as well as a couple Wemos D1 minis so I will have plenty to try come Tuesday if need be.

nstiesi, 2021-06-06:
Today is the first day that I have had it running for an extended period.  After multiple crashes in quick succession this morning, it seems to settle out to about once every 2-4 hours.  Not sure if this information is of any real value toward solving the root cause.

Thanks!

squishyu, 2021-06-07:
Am having the same issue, random reboots on every single WLED setup. Am using the the NodeMCU microcontroller and it gets quite frustrating since I can't figure out the source and it feels random. It feels like it reboots more frequently when running effects.

closed 2022-02-04 by blazoncek