wled/WLED #2019: WLED suddenly changes color setting

Ron0815, 2021-06-06:
WLED runs on an ESP8266 and is linked to the time-controlled switching of the outdoor-lighting through automation in Apple Homekit. As a result, the WLED should remain switched on continuously in warm-white for two hours. For no reason whatsoever, the WLED changes its color to red shortly after being switched on. What could be the reason? I soldered the circuit according to these instructions (circuit diagram further down on the page as follows): https://electric-junkie.de/2020/08/wled-quickstart/
What do I have to do? I would be pleased to get help, because my wife otherwise wants me to remove the installation. Best regards

sansillusion, 2021-06-07:
Does it stay white if you disable homekit automation and use the wled ui to
set to white ?
I suspect it is to do with homekit automation more than wled but I might be
wrong since I dont use Apple product so I know nothing of homekit.

Ron0815, 2021-06-07:
Yes, the status remains unaffected when I disable the automation in Homekit. But what is the reason? I did not alter the settings in homekit ...

huggy-d1, 2021-06-07:
Does the WLED UI show any time-based activities?  What preset runs at startup?

Ron0815, 2021-06-08:
No, there are no time-based activities. I don't know if a preset is running at start. In the config the preset 1 is at boot.

closed 2022-04-16 by stale[bot]