wled/WLED #2021: Boot issues when 74AHCT125 output is enabled on data pin

someguythatcodes, 2021-06-07:
Hi all,

I've designed a simple circuit (schematic attached) to use WLED  Wemos D1 mini board with a 74AHCT125 line driver and transistor to cut power to the LEDs when WLED is in off mode.

I'm currently facing an issue where if I power on or reset the D1 Mini, the D1 doesn't connect to the WIFI and the UI doesn't load. I've made some observations with only the 5V, GND and D4 pins connected on the D1 (basically eliminating the FET as the issue) and with no LED tape connected:

- If I disable the output enable of the 74AHCT125 it boots up correctly
- If I add a pull up or down to D4 (tried everything from 1k - 10k) I get the same problem
- If I remove the connection from the D4 to the 74AHCT125 and add a pull down to ground instead, I get the same issue.

That last point leads me to believe that the 74AHCT125 is pulling the line low during boot, which sends the ESP8266 into something other than normal running mode.

I've looked at a couple of other designs, and all have tied the 74AHCT125 output enable to ground, which has left me a bit stumped!

Has anyone else seen this?

Thanks in advance!

[image]

huggy-d1, 2021-06-07:
Take  a look at this : https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/ 
Scroll down to the **_Best Pins_** to Use section.  I think D4 is a problem pin.

Aircoookie, 2021-06-07:
D4/GPIO2 is the default though, even though it could be problematic, simply because it is the best pin for hardware driving (only GPIO1,2,3 on ESP8266, GPIO1 disables debugging and GPIO3 uses lots of memory).

I've never heard of this problem with a levelshifter, so this is a long shot, but could you try adding a 3.3kOhm or 4.7kOhm resistor between D4/the level shifter input and 3.3v? The reason is that the ESP will fail to boot if D4 is pulled low.

Aircoookie, 2021-06-07:
I should learn to read :) You already tried that. The 74AHCT125 should normally have a high impedance input, maybe it is faulty... Do you have a second one to test?

huggy-d1, 2021-06-07:
The reason I mentioned D4/GPIO2 is that is the on-board LED's pin as well.
All that said, when you have the D1 Mini removed from the circuit and connect via usb to a PC or USB power brick, does it work as expected with wifi and UI?

someguythatcodes, 2021-06-07:
> The reason I mentioned D4/GPIO2 is that is the on-board LED's pin as well.
> All that said, when you have the D1 Mini removed from the circuit and connect via usb to a PC or USB power brick, does it work as expected with wifi and UI?

D1 mini works fine when removed from the PCB and powered by a USB

> I should learn to read :) You already tried that. The 74AHCT125 should normally have a high impedance input, maybe it is faulty... Do you have a second one to test?

I was wondering this. Although I did a quick manual test (applied 0V/3.3V to it's input and ensured that 0V/5V came out the output) and it seems fine. Funnily enough I have had to replace the level shifter once already which resolved an output glitch issue. I'll try replacing again and report back

someguythatcodes, 2021-06-08:
Had some unexpected joy! I re-enabled the output on the 74AHCT125 by re-soldering the leg and it seems to be working perfectly. So hopefully just a bad solder joint, however I'll keep this running on the bench for a while to see if the problem returns

someguythatcodes, 2021-06-21:
Problem returned after a few days. I've tried running on GPIO1 instead of GPIO2, instead of the boot issue, I get lots of random white flashes over the entire LED strip and little to no control over it. Have also tried on several PCBs with new components each time

someguythatcodes, 2021-06-21:
I've also tried a 47 ohm and 220 ohm resister in line on the data line, which if if anything seemed to exaggerate the flashing problem!

someguythatcodes, 2021-06-30:
So it turns out my MOSFET low side switch was causing the upset, removed it from the equation and everything is working perfectly. Will be swapping it out for a MOSFET high side switch on the next revision. Thanks for all your help!

closed 2021-06-30 by someguythatcodes