wled/WLED #2022: Can’t connect to wled after attaching to my network

Kepler-11, 2021-06-09:
i was on the website 4.3.2.1 and i tried to connect my wled to my network and then when i go to 4.3.2.1 it fails to load and WLED-AP does not show up on my wifi list

Kepler-11, 2021-06-09:
also i bought this from a third party company that set it up for me so i don’t know how to fix this

sansillusion, 2021-06-09:
You need to find the IP of your wled from your router or a device scanner
once connected to your wifi. Or use the wled app on your phone to find it.

Kepler-11, 2021-06-09:
> You need to find the IP of your wled from your router or a device scanner once connected to your wifi. Or use the wled app on your phone to find it. Le mer. 9 juin 2021 3 h 28 a.m., Kepler-11 ***@***.***> a écrit :
> […](#)

Thank you, i downloaded a device scanner and got the ip, i can now change the leds from the app!

closed 2021-06-09 by Kepler-11