wled/WLED #2023: Buttons Macros wont trigger

Esquilo007, 2021-06-09:
**Describe the bug**
> The Macros for short press and long press button actions in the time and macros menu are switched
> When assigning a macro-id to the button actions the macros for double press as well as long press won't trigger.

**To Reproduce**
> Assign an id to the button actions in the time and macros menu and try the actions for short / long / double

**Expected behavior**
> Trigger Macro event

**WLED version**
 - Board: Wemos D1 mini
 - Version 0.12.1-b1
 - Format self-compiled, but nothing changed

**Additional context**
> I'm using 2 buttons

Esquilo007, 2021-06-09:
> What input PINs have you configured in WLED for the buttons?

GPIO4 (D2) and GPIO5 (D1) for the two buttons

huggy-d1, 2021-06-09:
Configured in the software, or using the WLED UI to select PIN assignments?  With 0.12+, the PIN assignments are done using the UI.  If you did that already, that's great.  Do you have a resistor to pull up/down the buttons so the voltage properly raises and falls?  I do not know if any pins in the D1 Mini have built-in pull-up/down resistors.

Esquilo007, 2021-06-10:
Done in the UI, that feature is great :)
I'm using two TP 223 touch sensors.
In the WLED UI both configured as normal push buttons, normally open.
I'm not 100% sure if using D1 and D2 on the D1 mini can cause problems

huggy-d1, 2021-06-10:
When you activate either touch sensor, what voltage is present at D1/D2?  What about when nothing is activating the sensors?
There are online resources showing how to configure and connect TP223's to ESP8266's.  I recommend checking them as a reference to help compare with what you have.  Should get you going fairly quickly.

blazoncek, 2021-06-10:
Have you tried with plain push-buttons?
If it works with plain (non-electronic/mechanical) buttons then check your touch buttons if they work correctly.

huggy-d1, 2021-06-10:
For that matter, how about trying simple dupont jumper wires from ground and +voltage to D1/D2 to verify they can work.

Esquilo007, 2021-06-11:
Thank you both, I'll try that.
Also: has anyone else with the newest version and using buttons also that behavior, that short trigger (push) and long trigger, setted in the UI is reversed? When configured as normal push buttons in the UI?

Esquilo007, 2021-06-11:
By the way, @blazoncek , thank you for that in my case very useful feature (multiple buttons)!

muhackl, 2021-07-14:
Same problem here. WEMOS D1 Mini. One plain push button. The one connector of the button to (with 10k pull-up-resistor) to 3V and to D3 as described in the picture here: https://github.com/Aircoookie/WLED/wiki/Macros.
The other connector of the button to GND. I configured three favorites and assigned them in the macros section to the three push patterns. Nothing happens, when I push the button accordingly. 
I also can not find the place in the UI where to configure the input pin (where should this be?) so I used the (hopefully) standard one.  
Is there any setting I missed? Something connected wrong?

blazoncek, 2021-07-14:
Button pin is defined in LED settings.

closed 2021-10-03 by blazoncek