wled/WLED #2027: New custom mapping not working as expected.

Chrisad21, 2021-06-14:
Hi , in previous version 0.11 i used this custom mapping settings and recompile the bin : 

`const uint16_t customMappingTable[] = {0, 1, 2, 3, 4, 5, 6, 7, 9, 11, 12, 14, 15, 16, 17, 21, 22, 24, 28, 32, 33, 34, 35, 
36, 38, 42, 48, 52, 54, 56, 59, 61, 62, 63, 64, 65, 67, 70, 71, 73, 74, 75, 78, 79, 80, 87, 90, 95, 97, 98, 99, 100, 106, 
108, 116, 118, 121, 123, 125, 126, 129, 131, 133, 134, 136, 139, 140, 142, 148, 152, 153, 158, 164, 165, 166, 167, 170, 
172, 174, 176, 182, 183, 187, 190, 196, 199, 204, 206, 207, 208, 209, 213, 214, 215, 216, 217, 218, 220, 224, 226, 227, 
230, 231, 232, 8, 10, 13, 18, 19, 20, 23, 25, 26, 27, 29, 30, 31, 37, 39, 40, 41, 43, 44, 45, 46, 47, 49, 50, 51, 53, 55, 
57, 58, 60, 66, 68, 69, 72, 76, 77, 81, 82, 83, 84, 85, 86, 88, 89, 91, 92, 93, 94, 96, 101, 102, 103, 104, 105, 107, 109, 
110, 111, 112, 113, 114, 115, 117, 119, 120, 122, 124, 127, 128, 130, 132, 135, 137, 138, 141, 143, 144, 145, 146, 147, 149, 
150, 151, 154, 155, 156, 157, 159, 160, 161, 162, 163, 168, 169, 171, 173, 175, 177, 178, 179, 180, 181, 184, 185, 186, 188, 189, 
191, 192, 193, 194, 195, 197, 198, 200, 201, 202, 203, 205, 210, 211, 212, 219, 221, 222, 223, 225, 228, 229};`

**This was worked fine, using these 2  segments (first : 0-104 and second : 104-232 ).**

In new version 0.12 i use the /edit page and ledmap.json to setup the exact same mapping : 

`{"map":[
  0, 1, 2, 3, 4, 5, 6, 7, 9, 11, 12, 14, 15, 16, 17, 21, 22, 24, 28, 32, 33, 34, 35, 
36, 38, 42, 48, 52, 54, 56, 59, 61, 62, 63, 64, 65, 67, 70, 71, 73, 74, 75, 78, 79, 80, 87, 90, 95, 97, 98, 99, 100, 106, 
108, 116, 118, 121, 123, 125, 126, 129, 131, 133, 134, 136, 139, 140, 142, 148, 152, 153, 158, 164, 165, 166, 167, 170, 
172, 174, 176, 182, 183, 187, 190, 196, 199, 204, 206, 207, 208, 209, 213, 214, 215, 216, 217, 218, 220, 224, 226, 227, 
230, 231, 232, 8, 10, 13, 18, 19, 20, 23, 25, 26, 27, 29, 30, 31, 37, 39, 40, 41, 43, 44, 45, 46, 47, 49, 50, 51, 53, 55, 
57, 58, 60, 66, 68, 69, 72, 76, 77, 81, 82, 83, 84, 85, 86, 88, 89, 91, 92, 93, 94, 96, 101, 102, 103, 104, 105, 107, 109, 
110, 111, 112, 113, 114, 115, 117, 119, 120, 122, 124, 127, 128, 130, 132, 135, 137, 138, 141, 143, 144, 145, 146, 147, 149, 
150, 151, 154, 155, 156, 157, 159, 160, 161, 162, 163, 168, 169, 171, 173, 175, 177, 178, 179, 180, 181, 184, 185, 186, 188, 189, 
191, 192, 193, 194, 195, 197, 198, 200, 201, 202, 203, 205, 210, 211, 212, 219, 221, 222, 223, 225, 228, 229]}`

**unfortunately it does not work as expected in new version using the same  2  segments (first : 0-104 and second : 104-232 )**.
the "missing"  led's from first segment are not lighting up from second segment. 

Second segment lights up only the led numbers which are grated than first segment  led's 

Hope the description to be clear. 

Best regards

huggy-d1, 2021-06-14:
Try the most recent firmware located here: https://github.com/Aircoookie/WLED/releases/tag/v0.12.1-b1
It supposedly fixes the problem you are having.

Chrisad21, 2021-06-14:
> 
> 
> Try the most recent firmware located here: https://github.com/Aircoookie/WLED/releases/tag/v0.12.1-b1
> It supposedly fixes the problem you are having.

Indeed Fixed!! Thanks for the quick response.

closed 2021-06-14 by Chrisad21

huggy-d1, 2021-06-14:
glad that fixed it !