wled/WLED #2028: Support for multiple relay board to control A/C lighting

cmgreenman, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
A means to drive multiple relays in order to control up to 8 individual dumb 120vac light strings.  

**Describe alternatives you've considered**
It might be possible using the RGBCW analog channels but I haven't tried (yet).  For my needs, just a 0v for off and +5v for relay on.  This will only provide 5 channels though.

**Additional context**
I found code for an esp8266 to drive the Sainsmart 8 channel relay (SSR) board over e1.31 at https://github.com/garymueller/ESP-131GPIO.  I thought it would be nice to have something similar running on an esp32 with ethernet and maintain all the wonderful wled goodness. 

Thank you for all of your hard work making WLED!!

huggy-d1, 2021-06-14:
If you configure WLED to drive analog LEDs using PWM to the various outputs, you could do as you suggest with 4 channels per WLED install, with 2 WLED analog installs to control 8 SSRs under PWM control.  
However, to do what you are thinking, I believe you probably want to add a usermod that communicates out of the WLED instance using E1.31 rather than into WLED using E1.31.

I guess I'm wondering what WLED would be doing to the various SSR outputs and how they would need to be configured to meet a proposed need.  Maybe I'm stuck with the proposed need since WLED is designed to control digital LEDs, but can treat a set of digital outputs as a single pixel for driving analog LEDs.

The problem is there are no 8-color LEDs currently on the market (as far as I know).  I've see what you mention with 2 types of white and 3 RGB channels, or 5 total colors for that single PWM controlled pixel.

Perhaps when in "analog mode", another setting for "# colors / pixel" could be made available, and then after a restart, each color could be given a pin a name and a pin # assignment.  Then after another reset, the colors / pin numbers would be displayed - in your case the color's could be SSR1 ... SSR8.  And of course, in analog mode, digital-only effects should automatically get filtered out.

Am I on a path that you think is helpful?

cmgreenman, 2021-06-16:
I think I figured it out actually.   I'm currently driving 8 SSRs with 1 wt32-eth01.  I set them up in the LED preferences using multiple single pin outputs.  Each works as a single analog LED channel and with segments I can control each ssr independently or as a group using segment 0.  The IO/pin I'm using for each of the outputs are:

1. IO2
2. IO4
3. IO12
4. IO14
5. IO17
6. IO5
7. IO15
8. IO32

I'm still testing and will be hooking up some actual A/C LED strings to the other side of the SSRs next.  I'm not holding my breath that dimming will work properly but on/off controlling is my main goal.   I also need to see how it plays with e1.31

blazoncek, 2021-06-18:
I have created a usermod (included with master now) that allows adding multiple relays to WLED.
You need to custom-build to activate it and be sure to add #defime MULTI_RELAY_MAX_RELAYS 8 (or -D MULTI_RELAY_MAX_RELAYS=8 for PIO) if you need 8 relays.

closed 2021-10-03 by blazoncek