wled/WLED #2029: i need help connect esp 8266 nodemcu esp-12 ch340 with ws2812b

tctmaster, 2021-06-15:
I connect VIN and GND to a common 5v6a power line and connect to leds without using a 4.7kohm divider between 3.3v and D4 since 4.7kohm is not available because of order and  wait shipping
Case 1: board is powered up and running online, but set LED SETUP pin = 2 (D4) or 4 (D2), nothing happens to LED, onboard LED stays blue
Case 2: Same as above but the board does not start and flashes, Led board flashes green once and then turns off

in my case is there any way to solve the problem from D2 and D4 pin and the circuit diagram from the shop I bought it says D4 pin is GPIO2 pin11

> [image]
> [image]

**WLED version**
 - Board: [default_envs = d1_mini because i don't know which one to choose for my esp 8266 nodemcu v3 esp-12e board]
 - Version [0.12.0]
 - Format [self-compiled]

tctmaster, 2021-06-15:
Can I ask is it a direct solder connection of 4.7k ohm from the 3.3v pin to the D4 pin?
[image]

huggy-d1, 2021-06-15:
Just connect up the LED strip data line directly to the D4 pin.  It will either work, or if the lead from the pin to the strip is too long, it won't.
If you have a level shifter to boost the 3.3V output from D4 to 5V for the LED strip, you insert that between the D4 output and the LED strip data connection.

tctmaster, 2021-06-15:
> Just connect up the LED strip data line directly to the D4 pin. It will either work, or if the lead from the pin to the strip is too long, it won't.
> If you have a level shifter to boost the 3.3V [...]

level shifter to increase 3.3V output from D4 to 5V for LED I don't use. I use 4.7k ohm connect D4 <> Led.can it work without 3.3v > 5v level booster?

huggy-d1, 2021-06-16:
Very often, the 3.3V output LED data pin provides enough voltage to drive the LED strip.  If the wire is small or long or both between the controller and LED strip, you should consider a level shifter to increase reliability.

tctmaster, 2021-06-17:
> Very often, the 3.3V output LED data pin provides enough voltage to drive the LED strip. If the wire is small or long or both between the controller and LED strip, you should consider a level shifter to increase reliability.
I'll take a look after the 4.7k ohm I ordered is delivered to try.
let me ask is my board esp 8266(esp-12e) nodemcu v3 ch340, which item should I choose default_envs as the most optimal item because I still don't know which one is the most suitable other than d1_mini

tctmaster, 2021-06-20:
can anyone tell me which entry should default_env be for esp-12e nodemcu 8266 v3

tctmaster, 2021-06-29:
> Very often, the 3.3V output LED data pin provides enough voltage to drive the LED strip. If the wire is small or long or both between the controller and LED strip, you should consider a level shifter to increase reliability.

i have used sn74hc125n and it is stable thanks, can i ask if i am going to have esp wroom 32d, do i need to flash bootloader first and then flash wled

closed 2022-02-04 by blazoncek