wled/WLED #2030: Possible to customize home page

Kepler-11, 2021-06-17:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

huggy-d1, 2021-06-17:
With a usermod, or simply customizing WLED and compiling / build WLED yourself, you can do whatever you like.  It is a somewhat involved process that I do not claim to understand.  However, the wled wiki has some information about it.

blazoncek, 2021-06-18:
It may be possible to _skin_ WLED in one of the next releases.
I have already implemented skinning (using CSS) in my experimental fork.
You can upload custom skin.css to the file system and it will override built-in CSS.

If you are inclined to completely override index.html, you can try to upload index.htm (or .html) to the file system. Don't forget your custom .js and .css.

tctmaster, 2021-06-20:
sorry to ask what impedance welding is the correct principle.
Case 1: solder 1 4.7k ohm pin from the 3.3v pin on the board and the other 1 pin to the D4 position pin.
Case 2: solder one 4.7k ohm pin to D4 and the other back to the DataInput wire of the led
which one is correct?

blazoncek, 2021-06-20:
> sorry to ask what impedance welding is the correct principle.
> Case 1: solder 1 4.7k ohm pin from the 3.3v pin on the board and the other 1 pin to the D4 position pin.
> Case 2: solder one 4.7k ohm pin to D4 and the other back to the DataInput wire of the led
> which one is correct?

Look at the Wiki page.

sanjulanipun, 2021-09-03:
> It may be possible to _skin_ WLED in one of the next releases.
> I have already implemented skinning (using CSS) in my experimental fork.
> You can upload custom skin.css to the file system and it will [...]

@blazoncek  **Can you Tell me you know how to covert index.html code as binary ?**

[image]

blazoncek, 2021-09-03:
https://kno.wled.ge/advanced/custom-features/

closed 2021-09-03 by blazoncek

sanjulanipun, 2021-09-03:
> https://kno.wled.ge/advanced/custom-features/

@blazoncek  
**ok, my problem is ok. but please i need to know how i can covert html code like this binary? this question for my knowlage. please tell me any tool or something how i can covert html code like this binary code. plz.**

spblat, 2025-08-09:
Is it feasible to present a custom page when the user connects to AP, such that a link in the custom page takes the user to the standard UI?