wled/WLED #2032: LEDs won't turn on when animated effect was selected for a lot of days. (Exoplanaton in post)

Nobodi99, 2021-06-19:
**Describe the bug**
I have a few instances of WLED running inside my appartement.
A lot of them are connected to power 24/7.
But LEDs are not on all the time. The board just has power so I can ask Alexa to turn it on if needed.
Most of the time I have some sort of animated effect running. (Like rainbow)
After maybe a few weeks, when I try to turn on the leds they don't turn on.
The webinterface says they are on but they arent't.
I have to manually reboot the board to get it working again.

I suspect it has to do with the millis() rollover but I'm not really sure where I would need to look inside the code to confirm that.
If anybody knows where I would need to look inside the code regarding the animations I would appreciate the help.

**To Reproduce**
Leave WLED running with the rainbow effect for a few weeks. LEDs don't have to be turned on.

**Expected behavior**
Whenever I try to turn on the LEDs they turn on without me restarting wled from the webinterface.

**WLED version**
 - Board: NodeMCU v2 ESP8266
 - Version: 0.11.1 but had the problem since 9.1
 - Format: binary and self-compiled (had both)

**Additional context**
If anyone can point me to the file where the animations are handled I could have a look myself.

Thank you for your help!

huggy-d1, 2021-06-21:
Does this happen with the most recent firmware build and release?  I say that because you posted an issue for an older and already superseded version - rather than posting in https://wled.discourse.group/

Nobodi99, 2021-06-22:
I just upgraded so it will take a while to confirm it. But there was nothing regarding that in the update notes so I just wanted to point it out here.

Aircoookie, 2022-04-16:
Please reopen if still an issue using v0.13.1 :)

closed 2022-04-16 by Aircoookie