wled/WLED #2034: Help build size firmware .irom0.text' will not fit in region `irom0_0_seg'

tctmaster, 2021-06-23:
I'm getting the same error now how do I fix esp-12e nodemcu v3 8266 with 4mb,I translate from ENG to my country language, i have edit rebuild npm for html_ui(261kb), html_other.h(743kb), html_settings.h(41kb) i 
Linking .pio\build\nodemcuv2\firmware.elf
c:/users/tctma/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\firmware.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1
thanks for help
[image]

closed 2021-06-25 by tctmaster

Sebbb, 2024-07-05:
Hey, I'm having the same issue, how did you solve it?

tctmaster, 2024-07-06:
> Hey, I'm having the same issue, how did you solve it?

You should reduce the image/similar file size because it exceeds the recorded memory because I replaced the background image without resizing and converting, so it gives an error

blazoncek, 2024-07-06:
It is static texts that do not fit.

tctmaster, 2024-07-08:
> It is static texts that do not fit.

For me, the problem is because the png file is not compressed, so it gives an error. Please check to see if the text file has also been converted to hex to reduce the size

tctmaster, 2024-07-08:
Have you converted html to binary array yet?
* Binary array for Web UI.
 * gzip is used for smaller size and improved speed.
 *
 * Please see https://github.com/Aircookie/WLED/wiki/Add-own-function#web-ui
 * to learn how to easily modify web UI source!