wled/WLED #2035: 0.12.1 beta crashes on TinyPico (ESP32 Pico D4) - but runs fine on 0.12.0

glowingkitty, 2021-06-23:
I compiled WLED via Platform.io in VS code (M1 Mac, MacOS 11.4).
The 0.12.1 beta firmware crashes on boot on the TinyPico and causes an infinite reboot loop.
Switched back to the code commit from the beginning of April - and that worked fine (no crash, no reboot).

Any idea what code change could have caused this crash?

huggy-d1, 2021-06-28:
In your TinyPico, does it also need a bootloader at 0x0000... and WLED loaded to 0x1000... like regular ESP32's?  I have no idea, but it is something to check and verify since you are loading firmware to a device nobody else seems to have mentioned having success with yet.

oliversettle, 2021-07-09:
Running into same issue with TinyPICO ESP32 
https://www.tinypico.com/
All firmware AFTER "WLED_0.11.1_ESP32_with_bootloader_beta" cause the wifi SSID is not seen and shows as hidden and manual config of this network does not work.
Used this resource 
https://wled.discourse.group/t/support-for-esp32-pico/265/14
Tried Manual build Via VSCODE but too much of noob to get working.
Any help would be amazing :)

Aircoookie, 2021-07-09:
Hey!
Sorry for the issue :)
Please try if the same issue happens if you use https://install.wled.me and select version 0.13.0-b0 in the dropdown.
If it still happens, please share the message the ESP prints to the serial monitor when bootlooping.

glowingkitty, 2021-07-09:
What’s new in 0.13.0? And any plans to include the sound reactive fork in the main repo?:)

oliversettle, 2021-07-09:
@Aircoookie Thank you so much for your time Christian your work is fantastic

TinyPICO ESP32-PICO-D4 (revision 1)
---------------------------------------------------------------------------------
USING - Web install https://install.wled.me

WLED SR 0.12.0 - ESP32 - YES  WiFi can connect - Installer UI reports 100% and done
WLED SR 0.12.1-b1 - ESP32 - No WiFi cannot connect - Installer UI reports 100% and done
WLED SR 0.13.0-b0 - ESP32 - No WiFi cannot connect - Installer UI reports 100% and done
----------------------------------------------------------------------------------
USING - ESPHome-Flasher-1.3.0-Windows-x64.exe

WLED SR 0.12.0 - ESP32 - Yes  WiFi can connect
WLED SR 0.13.0-b0 - ESP32 - No WiFi cannot connect - Boot loop - Shown Below cannot include full log as it overwrites when gets writing rom

VVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVVV
Using 'COM4' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-PICO-D4 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 94:B9:7E:82:4C:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz  
 
Using 'COM4' as serial port.
Unexpected error: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[22:50:50].pio/libdeps/esp32dev/LittleFS_esp32/src/lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
[22:50:50]E (59) esp_littlefs: mount failed,  (-84)
[22:50:50]E (60) esp_littlefs: Failed to initialize LittleFS
[22:50:50]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[22:50:50]Core 1 register dump:
[22:50:50]PC      : 0x40180e83  PS      : 0x00060130  A0      : 0x80180e54  A1      : 0x3ffb1d40  
[22:50:50]A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060520  A5      : 0x00000001  
[22:50:50]A6      : 0x00060520  A7      : 0x00000000  A8      : 0x80180e54  A9      : 0x3ffb1d10  
[22:50:50]A10     : 0x00000000  A11     : 0x00060523  A12     : 0x00060520  A13     : 0x3ffd40ac  
[22:50:50]A14     : 0x00000002  A15     : 0x3ffb1f00  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
[22:50:50]EXCVADDR: 0x0000004c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
[22:50:50]
[22:50:50]Backtrace: 0x40180e83:0x3ffb1d40 0x40180e51:0x3ffb1d70 0x401818aa:0x3ffb1da0 0x400dd9b6:0x3ffb1de0 0x400de85a:0x3ffb1e30 0x400e337b:0x3ffb1e90 0x400e35d2:0x3ffb1eb0 0x400fe545:0x3ffb1f20 0x400fe643:0x3ffb1f40 0x400ff996:0x3ffb1f90 0x4011e503:0x3ffb1fb0 0x4008a435:0x3ffb1fd0
[22:50:50]
[22:50:50]Rebooting...
[22:50:50]ets Jun  8 2016 00:22:57
[22:50:50]
[22:50:50]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:50:50]configsip: 188777542, SPIWP:0xee
[22:50:50]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:50:50]mode:DOUT, clock div:2
[22:50:50]load:0x3fff0018,len:4
[22:50:50]load:0x3fff001c,len:1044
[22:50:50]load:0x40078000,len:8896
[22:50:50]load:0x40080400,len:5828
[22:50:50]entry 0x400806ac
[22:50:51]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[22:50:51]Core 1 register dump:
[22:50:51]PC      : 0x40180e83  PS      : 0x00060530  A0      : 0x80180e54  A1      : 0x3ffb1b30  
[22:50:51]A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060920  A5      : 0x00000001  
[22:50:51]A6      : 0x00060920  A7      : 0x00000000  A8      : 0x80180e54  A9      : 0x3ffb1b00  
[22:50:51]A10     : 0x00000000  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000190  
[22:50:51]A14     : 0x00000000  A15     : 0x3ffcedec  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
[22:50:51]EXCVADDR: 0x0000004c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
[22:50:51]
[22:50:51]Backtrace: 0x40180e83:0x3ffb1b30 0x40180e51:0x3ffb1b60 0x401818aa:0x3ffb1b90 0x400dd9b6:0x3ffb1bd0 0x400de85a:0x3ffb1c20 0x400e337b:0x3ffb1c80 0x400e35d2:0x3ffb1ca0 0x400e99d6:0x3ffb1d10 0x400eac95:0x3ffb1ee0 0x400fe63b:0x3ffb1f40 0x400ff996:0x3ffb1f90 0x4011e503:0x3ffb1fb0 0x4008a435:0x3ffb1fd0
[22:50:51]
[22:50:51]Rebooting...
[22:50:51]ets Jun  8 2016 00:22:57
[22:50:51]
[22:50:51]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:50:51]configsip: 188777542, SPIWP:0xee
[22:50:51]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:50:51]mode:DOUT, clock div:2
[22:50:51]load:0x3fff0018,len:4
[22:50:51]load:0x3fff001c,len:1044
[22:50:51]load:0x40078000,len:8896
[22:50:51]load:0x40080400,len:5828
[22:50:51]entry 0x400806ac
[22:50:52]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[22:50:52]Core 1 register dump:
[22:50:52]PC      : 0x40180e83  PS      : 0x00060530  A0      : 0x80180e54  A1      : 0x3ffb1b30  
[22:50:52]A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060920  A5      : 0x00000001  
[22:50:52]A6      : 0x00060920  A7      : 0x00000000  A8      : 0x80180e54  A9      : 0x3ffb1b00  
[22:50:52]A10     : 0x00000000  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000190  
[22:50:52]A14     : 0x00000000  A15     : 0x3ffcedec  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
[22:50:52]EXCVADDR: 0x0000004c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
[22:50:52]
[22:50:52]Backtrace: 0x40180e83:0x3ffb1b30 0x40180e51:0x3ffb1b60 0x401818aa:0x3ffb1b90 0x400dd9b6:0x3ffb1bd0 0x400de85a:0x3ffb1c20 0x400e337b:0x3ffb1c80 0x400e35d2:0x3ffb1ca0 0x400e99d6:0x3ffb1d10 0x400eac95:0x3ffb1ee0 0x400fe63b:0x3ffb1f40 0x400ff996:0x3ffb1f90 0x4011e503:0x3ffb1fb0 0x4008a435:0x3ffb1fd0
[22:50:52]
[22:50:52]Rebooting...

oliversettle, 2021-07-11:
TinyPICO ESP32-PICO-D4 (revision 1)
[image]

esp32_bootloader_v2.bin
WLED_0.13.0-b2_ESP32.bin

ESPHome-Flasher-1.3.0-Windows-x64.exe - Bootloop Issue
flash_download_tool_3.8.8.exe - completes but no WLED AP SSID
https://install.wled.me  - completes but no WLED AP SSID

[image]

glowingkitty, 2021-07-13:
@Aircoookie as @oliversettle already mentioned - both the 0.13.0 b2 as well as the 0.12.1 b1 (sound reactive) end up in an infinite reboot loop after installing them via install.wled.me.
The 0.12.0 works fine. Any idea what changed since that version that could cause the crash?

srg74, 2021-07-18:
I'm trying to build binary for same board. Compilation of latest v0.13.b-2 is successful but board going in to loop.
Serial monitor output from PlatformIO:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4018777c: 0020c0a0 0b3088e4 77008000
Core 1 register dump:
PC      : 0x40187780  PS      : 0x00060b30  A0      : 0x80184a31  A1      : 0x3ffb1db0  
A2      : 0x3ff440c0  A3      : 0x3ff44000  A4      : 0x00060b20  A5      : 0x00000001  
A6      : 0x00060b20  A7      : 0x00000000  A8      : 0x80187744  A9      : 0x3ffb1d80  
A10     : 0x00000000  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40187780:0x3ffb1db0 0x40184a2e:0x3ffb1de0 0x40184e91:0x3ffb1e10 0x400de2ce:0x3ffb1e50 0x400de44e:0x3ffb1ea0 0x400e09e7:0x3ffb1ed0 0x400e0d02:0x3ffb1ef0 0x400f9b3d:0x3ffb1f20 0x400f9c49:0x3ffb1f40 0x400faf02:0x3ffb1f90 0x4011936a:0x3ffb1fb0 0x4008b082:0x3ffb1fd0

domisjustanumber, 2021-11-24:
*TL;DR*
I think I figured out what's going on here and you can get it to work if you jump through some hoops.

**Steps to get it working**
This is the process I found to get it working, but there may be an easier way (see my notes below).

1. Flash a release of WLED that includes the bootloader onto your board starting at address 0:
.\esptool.exe write_flash 0x0 ..\WLED_0.12.0_ESP32_with_bootloader.bin

This should work just fine and get it to come up. Connect to the WLED-AP and change the WiFi settings (or I suspect change anything at all - this seems to initialise the filesystem). If you don't do this, the next step won't work!

2. Once you've got a configured WLED install, upgrade to a newer version (in this case the latest soundreactive version).
.\esptool.exe -p COM6 write_flash 0x10000 ..\soundReactive_WLED_0.13.0-b4_ESP32.bin

### More info

**The controller**
I just ran into the same issue when using the [M5Stamp Pico](https://shop.m5stack.com/products/m5stamp-pico-diy-kit) board, which uses the same ESP32-PICO-D4 SoC. I suspect it also applies to the [M5Stack Atom Lite](https://shop.m5stack.com/products/atom-lite-esp32-development-kit) which appears to be exactly the same thing with an on-board USB controller.

**Specs on the ESP32-PICO-D4**
To clarify an error above - the ESP32-PICO-D4 is a full dual-core ESP32 packaged up in a SoC with some flash and other bits. (not to be confused with the single core ESP32-S2 or ESP32-C3 because geeze why not have more confusing product lineups).

> ESP32 contains two low-power Xtensa® 32-bit LX6 microprocessors. The internal memory includes:
> - 448 KB of ROM for booting and core functions.
> - 520 KB of on-chip SRAM for data and instructions.
> - 8 KB of SRAM in RTC, which is called RTC FAST Memory and can be used for data storage; it is accessed
> by the main CPU during RTC Boot from the Deep-sleep mode.
> - 8 KB of SRAM in RTC, which is called RTC SLOW Memory and can be accessed by the co-processor
> during the Deep-sleep mode.
> - 1 Kbit of eFuse: 256 bits are used for the system (MAC address and chip configuration) and the remaining
> 768 bits are reserved for customer applications, including flash-encryption and chip-ID.

[Full datasheet here](http://espressif.com/sites/default/files/documentation/esp32-pico-d4_datasheet_en.pdf)

**Possibly a filesystem setup issue**
I noticed that when flashing WLED-0.12.0 that there's a message in the log about the filesystem:

> Done! Flashing is complete!
> 
> Showing logs:
> [10:19:37].pio\libdeps\esp32dev\LittleFS_esp32\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
> [10:19:37]E (13) esp_littlefs: mount failed,  (-84)
> [10:19:37]E (13) esp_littlefs: Failed to initialize LittleFS
> [10:19:37]Ada
> 

And saw that this seems to be related to the bootloader not being installed properly, which is how I came to the finding on how to make this work.

I'm still seeing some instability and crashes when I change the sound input pins, but that is a separate issue and may be something to do with me rather than the code.

Hopefully that helps direct someone that understands more about the flashing process to iron out that bug in the super-cool web installer!

domisjustanumber, 2022-01-10:
While doing some testing with the sound reactive branch, I found out that the default LED pin seems to be what causes the loop crash on the M5 Stamp Pico (also based on the ESP32-PICO-D4).

If you change the default LED pin (I set it to 27 for the on-board SK6812 on the M5Stamp), it will flash and boot properly via PlatformIO. If you try making a new env or just add `-D LEDPIN=xx` to the `build_flags` where xx is your LED pin, it should (might?) work for the TinyPICO as well.

I checked a new env for this into the soundreactive branch specifically for the M5Stamp Pico's pins, so you can see what I'm talking about: https://github.com/atuline/WLED/pull/151

blazoncek, 2022-10-10:
As 0.14 will add (at least partial) support for ESP32-S2, -S3 and -C3 I am closing this issue as irrelevant as it was opened for 0.12

closed 2022-10-10 by blazoncek

glowingkitty, 2022-10-10:
Is there an overview somewhere about what features don’t work yet reliable on the S2, S3 and C3?

blazoncek, 2022-10-10:
No, check (and follow) Discord beta channels.