wled/WLED #2036: N-Channel-Mosfet instead Relay needs Data-Line pulled high when WLED switched off

Saschko2000, 2021-06-24:
**Is your feature request related to a problem? Please describe.**
I have not enough space for a relay. And I tested to switch the Ground of the stripe through a n-channel mosfet.
It works perfectly if you switch the data.line of the stripe high wile the stripe should be off.

**Describe the solution you'd like**
An option to pull the data-lines high while off.

**Describe alternatives you've considered**
I would try to programm it by myselfe, but i don't know if my skills are good enough :-(

**Additional context**
I like the project very much :-)

Thank you for your ideas for making WLED better!

smartroad, 2021-06-25:
As a temporary stop gap, could you use a resistor to pull the pin high? Just connect between the pin and 3v?

Saschko2000, 2021-06-25:
> As a temporary stop gap, could you use a resistor to pull the pin high? Just connect between the pin and 3v?

Thank you for your Idea :-)

But I think I made a mistake. I did my tests with an Arduino Nano. And there ist the dataline 5V. I think i have to change my layout and use a p-channel-mosfet to switch the 5v line.

closed 2021-06-25 by Saschko2000