wled/WLED #2039: Use digital inputs of ESP (Controller) for loading presets

xozy22, 2021-06-24:
It would be nice to use additional DIs to load WLED presets.
This feature could enable the Microcontroler for standalone applications without having an additional raspberry pi sending UDP messages or something.
For example garage entrance lights, like in this video:
https://youtu.be/M8xCgQUihd8

blazoncek, 2021-06-28:
Use multiple buttons feature in current master.
It was implemented in #1977 .

xozy22, 2021-06-29:
So I am correct in assuming that only D3 is used. This then reacts to short press, long press and double press?
Thats not ideal for usage with magnetic contacts, but ok. I will find a solution.
Thanks 👍

blazoncek, 2021-06-29:
A switch is a switch, mechanical or magnetic.
Switches do not have short press, long press or double press states, but react on transition edge (low to high or high to low). You still use fields for short press and long press to assign action for transition edge.

closed 2021-10-03 by blazoncek