wled/WLED #2040: OTA update for 1M flash size?

golddragon007, 2021-06-26:
**Is your feature request related to a problem? Please describe.**
I have several magic home spi devices with 1M flash size. This device is pretty much hard to flash and they are also stuck to somewhere on a surface so I want to avoid flashing it with a cable.

**Describe the solution you'd like**
OTA update, one step or two-step like tasmota has.

**Describe alternatives you've considered**
-

**Additional context**
-

blazoncek, 2021-06-28:
Not possible. Not even with workarounds since even the smallest WLED image is still too big.

huggy-d1, 2021-06-28:
If you go back to prior to v0.11, I think it will fit and allow OTA updates, but once you get 0.11 installed, I think that's as far as it goes.
To go further requires hardware modification to your device.  Find the 1M flash chip.  Search YT for 4M flash chip upgrades for ESP8266 (like SONOFFs), order new flash and install - assuming you feel comfortable doing that.  Or, ask a friend with better soldering skills / equipment to help.

golddragon007, 2021-06-28:
Hmmm... so then I need to remove the ESP shield, remove the flash chip, install probably a W25Q32FVSSIG flash chip and install back the ESP shield. I just hope there will be enough space under the shield for this.

golddragon007, 2021-08-04:
[image]
Haha I can't find here the flash chip :D great...

huggy-d1, 2021-08-04:
That has the ESP8285 which has 1MB flash built into the microcontroller chip. So, it’s there but hidden from prying eyes.

golddragon007, 2021-08-04:
That means that I can't upgrade it. At least not in an easy way...

Leo-PL, 2021-08-20:
Nowadays, you can buy 2MB ESP8285 chips and modules containing it. Just look for this in the item description. My SP501E controller had 2MB ESP8285 from the get-go.

golddragon007, 2021-08-20:
They don't write this information there, I've just bought now SP501E, and I've just given up trying to flash them, For some reason, I can't make it. I've always got permission denied when I try to run the esptool.py chip_id command from admin cmd... I might try tomorrow to solder the pins there, but I don't see too much success rate right now with it...

golddragon007, 2021-08-20:
Oh and I HOPE that I've got 2M version, otherwise I will have a bad day...

Phara0h, 2021-09-20:
Would love to have an option for this even if it requires no ui (configure via http rest) and all other addons turned off besides e1.31

blazoncek, 2021-09-21:
> Would love to have an option for this even if it requires no ui (configure via http rest) and all other addons turned off besides e1.31

That would mean strip 90% of WLED code. 🤣 
You may be better off with some other controller.

Phara0h, 2021-09-22:
> > Would love to have an option for this even if it requires no ui (configure via http rest) and all other addons turned off besides e1.31
> 
> That would mean strip 90% of WLED code. 🤣
> You may be better off with some other controller.

Oh deff would easier to just use a better controller however not something that is changeable in RGB light bulbs.

blazoncek, 2021-09-22:
I personally use quite a few ESP01 modules and trust me I'd like to see a possibility to OTA those.
We tried many options but WLED just became too big for any solution.
OTA on 1MB chips is unachievable at this point.
Closing.

closed 2021-09-22 by blazoncek

blazoncek, 2021-11-28:
Good news on OTA update for ESP01 or other 1MB flash units.

@DrTFav (tony) on Discord has devised a method (2 step) that allows updating 1M units with latest version using OTA update. It is a bit involved and consists of the following:

Prepare minimal updater:

1. clone https://github.com/Jason2866/ESP8266MinimalHTTPUpdater/
2. edit main.ino and replace entire content with
```c++
#include <ESP8266WiFi.h>
#include <ArduinoOTA.h>

#define STASSID "YOURSSID"
#define STAPSK  "YOURPASS"

char ssid[] = STASSID;
char password[] = STAPSK;

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.hostname("ESP-MinUpdater");
  WiFi.setOutputPower(17);
  WiFi.begin(ssid, password);
  ArduinoOTA.begin();
}

void loop() {
  ArduinoOTA.handle();
}
```
3. build it using PIO
4. gzip updater.bin and copy it to a safe location (as updater.bin.gz)

Prepare appropriate WLED binary:

1. build WLED firmware by extending `env:esp01_1m_full` in PIO but using such overrides that the code size is less than 715000 bytes  (you may use: -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_LOXONE -D WLED_DISABLE_MQTT -D WLED_DISABLE_WEBSOCKETS), make sure OTA is **enabled**
2. upload this firmware using USB (only needed first time)
3. configure WLED as usual

When you are ready to update your ESP01 OTA do the following:

1. use espota.py tool to upload minimal updater to ESP01
```
python3 espota.py -i 192.168.x.x -p 8266 -f /path/to/updater.bin.gz 
```
2. wait for ESP to restart and connect to your WiFi (assuming you have correctly entered SSID and password in main.ino)
3. use espota.py tool to upload new WLED firmware
```
python3 espota.py -i 192.168.x.x -p 8266 -f /path/to/wled-binary.bin 
```

If the update fails it may mean your binary is too big. You can try to gzip it (I did not have success though).
I also noticed that WLED _forgot_ my WiFi credentials during the update but retained all other settings.

tony-fav, 2021-11-28:
Thanks for the write up @blazoncek! This methods works pretty well, all; but, I would suggest practicing the upgrade loop on something easy to serial flash like an ESP01. I've used this method to put WLED on some RGB PWM bulbs for those sweet effects 😁, but each time I compile a new WLED binary or minimal updater binary, I make sure to practice on something easier to serial flash if something breaks (sizes don't work for some reason).

blazoncek, 2021-12-11:
I have prepared a bash script if anyone wants to bulk update many ESP01's (or just one).

```bash
#!/bin/bash
MINPATH=~/Documents/Arduino/ota-mini/build_output/firmware
WLEDPATH=~/Documents/Arduino/WLED/build_output/firmware
ESPPATH=~/.platformio/packages/framework-arduinoespressif32/tools

update_one() {
  ping -c 1 -t 2 $1 >/dev/null
  PINGRESULT=$?
  if [ $PINGRESULT -eq 0 ]; then
    echo Updating $1
    curl -s http://$1/presets.json -o presets-tmp.json >/dev/null
    curl -s http://$1/cfg.json -o cfg-tmp.json >/dev/null
    python3 $ESPPATH/espota.py -i $1 -p 8266 -f $MINPATH/updater.bin.gz
    sleep 15
    python3 $ESPPATH/espota.py -i $1 -p 8266 -f $WLEDPATH/esp01_1m_ota.bin
    sleep 15
    curl -s -F "data=@cfg-tmp.json;filename=/cfg.json" http://$1/upload >/dev/null
    curl -s -F "data=@presets-tmp.json;filename=/presets.json" http://$1/upload >/dev/null
    curl -s http://$1/reset >/dev/null
    rm presets-tmp.json
    rm cfg-tmp.json
    return 0
  else
    echo Missing $1
  fi
  return 1
}

update_one 192.168.x.x
update_one 192.168.x.x
```

Replace WLEDPATH, MINPATH and ESPPATH with paths for your environment. You will also need to clone Minimal Updater as noted above and have it compiled with your WiFi credentials.
The same goes for WLED.
Use `-D CLIENT_SSID='"yourssid"'` and `-D CLIENT_PASS='"yourpassword"'` in your build environment (assumed `[env:esp01_1m_ota]` for the script above) for WLED (in platformio.ini or platformio_override.ini).

Imperattor, 2022-02-23:
> [image] Ха-ха, я не могу найти здесь флэш-чип :D отлично...

Could you please tell me what kind of board this is?