wled/WLED #2041: Segment Effect Offset

Sturm2002, 2021-06-26:
For each segment a offset from which the effects start (not the segment itself), could help to sync the leds even more

**Describe the solution you'd like**
Maybe another input field for each segment, to input a number (offset)

**Additional context**
Imagine having Leds arranged in a circle, the segment starts Top-Left (NW) (Picture 1), if you now give it an effect-offset (let's say 10 Leds), it would look like the segment would start from North (Picture 2)

Picture 1 :
[image]

Picture 2 :
[image]

blazoncek, 2021-06-28:
This can be done using LED map.
Create a ledmap.json file and upload it to the file system on /edit.
Look into #1738 and Wiki.

Sturm2002, 2021-06-28:
oh, whoops i kinda missed that :D
thank you

closed 2021-06-28 by Sturm2002