wled/WLED #2042: ESP32 Wroom 32D led strip only white

MadaCrX, 2021-06-27:
Hello guys. I bought from aliexpress 3 pieces of ESP32 Wroom 32D..and i wanted to flash WLED on ESP32 like a guy from youtube did .. and i can't make it work.. The led strip light only on full white collor.. I soldered the wires exactly where they have to be soldered, i did no mistakes. I have ws2812B led strip only 60 leds on that led strip piece that i'm using from test my power supply it's 5V with 15A . I did everything like that guy from youtube did...and for me don't work..but for him works. I just downloaded " ESPHome-Flasher-1.3.0-Windows-x64 " to flash the ESP32 and i flashed it with this firmware " WLED_0.12.0_ESP32 " Please don't take me hard..i'm newbie :D .  Thank you very much !! This is the guy who did the tutorial ...and for him...everything works like a charm...but for me..no :( .. [https://youtu.be/j15pjU3YZfg?t=656](url)

Sturm2002, 2021-06-28:
I think you don't have the right driver, so your pc doesn't recognize the ESP32

try to install this driver if you're on windows:  **CP210x Universal Windows Driver**

https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

then try again

huggy-d1, 2021-06-28:
Did you install the bootloader to 0x0000... and WLED to 0x1000... like in the WLED wiki for ESP32?

closed 2022-04-16 by Aircoookie