wled/WLED #2047: One in three LEDs have the correct colour

lzmartinico, 2021-06-30:
**Describe the bug**
On a newly purchased WS2812b 5 meter light strip, setting any colour other than white through the pallet instrument results in 1/3 LEDs getting the correct colour, with opposite colours in the RGB spectrum. e.g. if I set my colour to Green, solid, the LEDs will reproduce the pattern of Green, Red, Blue, GRB etc. If I make a segment with spacing of 3, they will all feature the correct colour.

**To Reproduce**
Select a non-white colour from the palette, choose Color 1 (or any other scheme featuring a colour). The LEDs will turn on the right colour on the first (and every third LED) and other colours for the remaining groups of two

**Expected behavior**
All LEDs should have the same colour

**WLED version**
 - Board: ESP8266
 - Version: 0.12, build 2104020
 - Format: Binary, pre-compiled

Thank you for your help!

Aircoookie, 2021-06-30:
Hi!
Sounds like you might be in RGBW mode for some reason.
Make sure that your LED type in LED settings -> Hardware setup is set to `WS2812x`, not `SK6812` :)

lzmartinico, 2021-06-30:
Hey, I am pretty sure that I have the right device selected! I also tried switching between the various colour order options (RGB, GRB etc) and all that does is change the order of the correctly coloured LEDs (as you'd expect)

lzmartinico, 2021-06-30:
Ah, very interestingly, I actually switched to SK6812 and it works!

Aircoookie, 2021-06-30:
Very interesting. I know that sometimes SK6812 is sold as WS2812B, but not 4-channel chips with 3-channel LEDs

closed 2021-06-30 by Aircoookie