wled/WLED #2050: Hardware buttons config getting wrong after re-opening "config" page

ITqua, 2021-07-01:
** Issue description **
This issue related to pull #1977 (Multi-button support).
Button 0 behavior is getting reset / wrong after re-opening "config" page.
Settings works correct after saving and until config page is opened again.

**To Reproduce**
1. Open Config -> LED & Hardware setup (http://your-wled-ip/settings/leds)

2. Set buttons config [as following](http://i.piccy.info/i9/5a66cf2586eb72b7a19f9842d39c3b68/1625139107/5498/1431576/89043Screenshot_2.png):
 Button 0 pin: Switch
 Button 1 pin: Push button
 Button 2 pin: Push button
 Button 3 pin: Push button

3. Save changes. They should work as expected.

4. Now open the same page again (or open any config page and then go to http://your-wled-ip/settings/leds)

5. The button config [is wrong](http://i.piccy.info/i9/2bc70b9761fdebce89954d71b40ea03f/1625139265/9187/1431576/Screenshot_3.png):

 Button 0 pin: Switch inverted (wrong)
 Button 1 pin: Push button (correct)
 Button 2 pin: Push button (correct)
 Button 3 pin: Push button (correct)

**Expected behavior**
Button 0 config should remain just as it stated in settings.

**WLED version**
 - Board: ESP32 devkit V4
 - Version: v 0.12.1-b1 "Hikari", build: 2106250
 - Format: self-compiled

**Additional context**
Please see the demo video showing the issue:
https://youtu.be/r5XiEcg-7Es

ITqua, 2021-07-04:
@Aircoookie The bug is also presents in latest 0.13.0-b0 beta version.

blazoncek, 2021-07-12:
> @Aircoookie The bug is also presents in latest 0.13.0-b0 beta version.

Impossible since 0.13 does not have "Switch inverted". :) Refresh your browser cache.
This was fixed in #2011

ITqua, 2021-07-12:
> > @Aircoookie The bug is also presents in latest 0.13.0-b0 beta version.
> 
> Impossible since 0.13 does not have "Switch inverted". :) Refresh your browser cache.
> This was fixed in #2011

You are right. Could you please confirm that in latest version Switch is working correctly?

In my case (board was reflashed with latest build, cache cleared):

first time Switch working as it supposed, after second switch release it starts working in inverted mode.
The correct switch behavior is set when PIR option selected (but Switch is connected to pin instead of PIR detector)

blazoncek, 2021-07-12:
Switch reacts on a transition from high to low or the other way arround. You can swap presets if the order is not correct in your wiring.

closed 2021-10-03 by blazoncek